EIPStackGroup/OpENer #254: Async instead of timeout ?

yanzixiang, 2019-08-31:
I use [arduino-esp32](https://github.com/espressif/arduino-esp32),[AsyncTCP](https://github.com/me-no-dev/AsyncTCP),[AsyncUDP](https://github.com/espressif/arduino-esp32/tree/master/libraries/AsyncUDP),

It is very fast.

It seems that the code here costs a lot of time waitting for the packets to come,
Is there an ASYNC way to save the time ?
```c
g_time_value.tv_sec = 0;
  g_time_value.tv_usec = (
    g_network_status.elapsed_time < kOpenerTimerTickInMilliSeconds ?
    kOpenerTimerTickInMilliSeconds - g_network_status.elapsed_time : 0)
                         * 1000; /* 10 ms */
  int ready_socket = select(highest_socket_handle + 1, &read_socket, 0, 0,
                            &g_time_value);
```

CapXilinx, 2019-09-02:
Feel free to tryout. From my experience the select loop performs fast enough, as it only has to handle at max 23 sockets or so? So this is not a priority on my todo list. You could also try to change the select loop to libevent calls (https://libevent.org/)
If you do all your tasks before the select call, it just waits the time difference to 10ms. If you need to do more in parallel, you could put OpENer into a separate thread.

yanzixiang, 2019-09-05:
I think libevent has NO support for FreeRTOS.

closed 2019-09-05 by yanzixiang