EIPStackGroup/OpENer #255: NO T->O PACKET SENT OUT

yanzixiang, 2019-08-31:
[20190831_0002.zip](https://github.com/EIPStackGroup/OpENer/files/3562533/20190831_0002.zip)

wireshark just get the packet from my pc to my board,
there is no packet from my board to my pc to update the values,
please help.

22:09:31.489 -> [E][generic_networkhandler.c:296] CheckAndHandleTcpListenerSocket(): networkhandler: new TCP connection
22:09:31.489 -> [E][generic_networkhandler.c:320] CheckAndHandleTcpListenerSocket(): >>> network handler: accepting new TCP socket: 57 
22:09:31.523 -> [E][generic_networkhandler.c:338] CheckAndHandleTcpListenerSocket(): New highest socket: 57
22:09:31.523 -> [E][generic_networkhandler.c:343] CheckAndHandleTcpListenerSocket(): networkhandler: opened new TCP connection on fd 57
22:09:31.523 -> [E][generic_networkhandler.c:946] CheckAndHandleConsumingUdpSockets(): start
22:09:31.626 -> [E][generic_networkhandler.c:946] CheckAndHandleConsumingUdpSockets(): start

22:09:31.626 -> [E][generic_networkhandler.c:614] HandleDataOnTcpSocket(): Entering HandleDataOnTcpSocket for socket: 57
22:09:31.626 -> [E][generic_networkhandler.c:744] HandleDataOnTcpSocket(): Data received on tcp:
22:09:31.626 -> [E][encap.c:158] HandleReceivedExplictTcpData(): Handles data for TCP socket: 57
22:09:31.626 -> [E][encap.c:201] HandleReceivedExplictTcpData(): Register session
22:09:31.626 -> [E][socket_timer.c:14] SocketTimerSetSocket(): Adds socket 57 to socket timers
22:09:31.626 -> [E][socket_timer.c:21] SocketTimerSetLastUpdate(): Sets time stamp for socket 57
22:09:31.659 -> [E][socket_timer.c:21] SocketTimerSetLastUpdate(): Sets time stamp for socket 57
22:09:31.659 -> [E][generic_networkhandler.c:782] HandleDataOnTcpSocket(): TCP reply sent:
22:09:31.659 -> [E][socket_timer.c:21] SocketTimerSetLastUpdate(): Sets time stamp for socket 57

22:09:31.728 -> [E][generic_networkhandler.c:946] CheckAndHandleConsumingUdpSockets(): start
22:09:31.728 -> [E][generic_networkhandler.c:614] HandleDataOnTcpSocket(): Entering HandleDataOnTcpSocket for socket: 57
22:09:31.728 -> [E][generic_networkhandler.c:744] HandleDataOnTcpSocket(): Data received on tcp:
22:09:31.728 -> [E][encap.c:158] HandleReceivedExplictTcpData(): Handles data for TCP socket: 57

22:09:31.728 -> [E][encap.c:214] HandleReceivedExplictTcpData(): Send Request/Reply Data

22:09:31.728 -> [E][cpf.c:246] CreateCommonPacketFormatStructure(): Sockaddr type id: 8001

22:09:31.728 -> [E][cipmessagerouter.c:183] NotifyMessageRouter(): NotifyMessageRouter: routing unconnected message
22:09:31.763 -> [E][cipmessagerouter.c:219] NotifyMessageRouter(): NotifyMessageRouter: calling notify function of class 'connection manager'
22:09:31.763 -> [E][cipmessagerouter.c:137] GetCipInstance(): start instance_number:1
22:09:31.763 -> [E][cipcommon.c:88] NotifyClass(): notify: found instance 1
22:09:31.763 -> 
22:09:31.763 -> [E][cipcommon.c:99] NotifyClass(): notify: calling ForwardOpen service
22:09:31.763 -> 
22:09:31.763 -> [E][cipconnectionmanager.c:509] ForwardOpen(): start
22:09:31.763 -> [E][cipconnectionobject.c:224] ConnectionObjectSetState(): CIP_CONNECTION_OBJECT_STATE_NON_EXISTENT
22:09:31.832 -> [E][cipconnectionobject.c:157] ConnectionObjectInitializeFromMessage(): ForwardOpen: ConConnID 124350898, ProdConnID 124350899, ConnSerNo 30360
22:09:31.832 -> 
22:09:31.832 -> [E][cipconnectionmanager.c:553] ForwardOpen(): We have a Non-Null request
22:09:31.832 -> 
22:09:31.832 -> [E][cipconnectionmanager.c:562] ForwardOpen(): We have a Non-Matching request
22:09:31.832 -> 
22:09:31.832 -> [E][cipconnectionmanager.c:569] ForwardOpen(): end
22:09:31.832 -> [E][cipconnectionmanager.c:406] HandleNonNullNonMatchingForwardOpenRequest(): start
22:09:31.866 -> [E][cipconnectionobject.c:353] ConnectionObjectGetTransportClassTriggerProductionTrigger(): start
22:09:31.866 -> [E][cipconnectionmanager.c:1084] ParseConnectionPath(): start
22:09:31.866 -> [E][cipconnectionmanager.c:1088] ParseConnectionPath(): remaining_path : 4
22:09:31.866 -> [E][cipconnectionobject.c:591] ConnectionObjectSetProductionInhibitTime(): production_inhibit_time :256
22:09:31.866 -> [E][cipconnectionmanager.c:1116] ParseConnectionPath(): kSegmentTypeLogicalSegment
22:09:31.866 -> [E][cipconnectionobject.c:353] ConnectionObjectGetTransportClassTriggerProductionTrigger(): start
22:09:31.901 -> [E][cipepath.c:333] CipEpathGetLogicalValue(): start
22:09:31.901 -> [E][cipepath.c:336] CipEpathGetLogicalValue(): logical_format : 0
22:09:31.901 -> [E][cipconnectionmanager.c:1191] ParseConnectionPath(): class_id : 4
22:09:31.901 -> [E][cipconnectionmanager.c:1210] ParseConnectionPath(): classid 4 (assembly)
22:09:31.901 -> 
22:09:31.901 -> [E][cipconnectionmanager.c:1217] ParseConnectionPath(): remaining_path : 4
22:09:31.901 -> [E][cipconnectionmanager.c:1219] ParseConnectionPath(): remaining_path : 3
22:09:31.901 -> [E][cipepath.c:333] CipEpathGetLogicalValue(): start
22:09:31.901 -> [E][cipepath.c:336] CipEpathGetLogicalValue(): logical_format : 0
22:09:31.934 -> [E][cipconnectionmanager.c:1227] ParseConnectionPath(): Configuration instance id 1
22:09:31.934 -> 
22:09:31.934 -> [E][cipmessagerouter.c:137] GetCipInstance(): start instance_number:1
22:09:31.934 -> [E][cipconnectionmanager.c:1236] ParseConnectionPath(): remaining_path : 3
22:09:31.934 -> [E][cipconnectionmanager.c:1238] ParseConnectionPath(): remaining_path : 2
22:09:31.934 -> [E][cipconnectionobject.c:388] ConnectionObjectGetTransportClassTriggerTransportClass(): start
[... 1089 lines cut ...]
22:10:05.152 -> 
22:10:05.152 -> [E][cipconnectionobject.c:388] ConnectionObjectGetTransportClassTriggerTransportClass(): start
22:10:05.220 -> [E][cipconnectionobject.c:415] ConnectionObjectGetTransportClassTriggerTransportClass(): transport_class_trigger : 1
22:10:05.220 -> [E][cipioconnection.c:930] HandleReceivedIoConnectionData(): data length after sequence count: 36
22:10:05.220 -> 
22:10:05.220 -> [E][cipioconnection.c:935] HandleReceivedIoConnectionData(): Run/Idle handler: 0x1
22:10:05.220 -> [E][cipidentity.c:186] CipIdentitySetExtendedDeviceStatus(): Setting extended status: 60
22:10:05.220 -> [E][YZXDCPServer.cpp:357] AfterAssemblyDataReceived(): AfterAssemblyDataReceived on Core 1 Times: 66 
22:10:05.220 -> [E][YZXDCPServer.cpp:366] AfterAssemblyDataReceived(): g_assembly_data064[0]: 64 ----- g_assembly_data096[0]: 65
22:10:05.458 -> [E][generic_networkhandler.c:946] CheckAndHandleConsumingUdpSockets(): start
22:10:05.458 -> [E][generic_networkhandler.c:614] HandleDataOnTcpSocket(): Entering HandleDataOnTcpSocket for socket: 57
22:10:05.458 -> 
22:10:05.458 -> [E][generic_networkhandler.c:655] HandleDataOnTcpSocket(): networkhandler: error on recv: 104 - Connection reset by peer
22:10:05.458 -> 
22:10:05.458 -> [E][generic_networkhandler.c:1011] CloseSocket(): networkhandler: closing socket 57
22:10:05.458 -> 
22:10:05.458 -> [E][networkhandler.c:44] CloseSocketPlatform(): Could not close socket 57 - Error Code: 128 - Socket is not connected
22:10:05.492 -> 
22:10:05.492 -> [E][generic_networkhandler.c:1017] CloseSocket(): networkhandler: closing socket done 57
22:10:05.492 -> 
22:10:05.492 -> [E][encap.c:835] RemoveSession(): encap.c: Removing session
22:10:05.492 -> 
22:10:05.492 -> [E][cipconnectionobject.c:388] ConnectionObjectGetTransportClassTriggerTransportClass(): start
22:10:05.492 -> [E][cipconnectionobject.c:415] ConnectionObjectGetTransportClassTriggerTransportClass(): transport_class_trigger : 1
22:10:05.492 -> [E][encap.c:843] RemoveSession(): encap.c: Session removed

CapXilinx, 2019-09-02:
What is happening exactly? Does your scanner have a timeout? Because it usually should only try to for a period of 10 seconds at first. Your trace shows, that the O to T connection is up for 30 seconds.

yanzixiang, 2019-09-02:
> What is happening exactly? Does your scanner have a timeout? Because it usually should only try to for a period of 10 seconds at first. Your trace shows, that the O to T connection is up for 30 seconds.

I ended the program runs as the scanner.

> If there is no timeout, this would hint, that you actually get packets from OpENer, but you don't see them in Wireshark

I don't think so ,because  the scanner is my pc which is just the Wireshark runs on. 
It seems that OpENer failed to create the UDP socket of T->O,
And just called CheckAndHandleConsumingUdpSockets() when it got the O->T packets.
It seems that it is the LWIP problem, 
I am trying to figure out how to fix.

yanzixiang, 2019-09-02:
ManageConnections() faild to trigger.
fixed.
thanks.

closed 2019-09-02 by yanzixiang