EIPStackGroup/OpENer #256: kConnectionManagerExtendedStatusCodeErrorOwnershipConflict when reconnect

yanzixiang, 2019-09-02:
[20190902_0003.zip](https://github.com/EIPStackGroup/OpENer/files/3565931/20190902_0003.zip)

19:47:45.356 -> [E][generic_networkhandler.c:296] CheckAndHandleTcpListenerSocket(): networkhandler: new TCP connection
19:47:45.356 -> [E][generic_networkhandler.c:320] CheckAndHandleTcpListenerSocket(): >>> network handler: accepting new TCP socket: 57 
19:47:45.356 -> [E][generic_networkhandler.c:338] CheckAndHandleTcpListenerSocket(): New highest socket: 57
19:47:45.356 -> [E][generic_networkhandler.c:343] CheckAndHandleTcpListenerSocket(): networkhandler: opened new TCP connection on fd 57
19:47:45.356 -> [E][generic_networkhandler.c:949] CheckAndHandleConsumingUdpSockets(): start
19:47:45.390 -> [E][generic_networkhandler.c:949] CheckAndHandleConsumingUdpSockets(): start
19:47:45.390 -> [E][generic_networkhandler.c:614] HandleDataOnTcpSocket(): Entering HandleDataOnTcpSocket for socket: 57
19:47:45.390 -> [E][generic_networkhandler.c:745] HandleDataOnTcpSocket(): Data received on tcp:
19:47:45.390 -> [E][encap.c:158] HandleReceivedExplictTcpData(): Handles data for TCP socket: 57
19:47:45.390 -> [E][encap.c:201] HandleReceivedExplictTcpData(): Register session
19:47:45.390 -> [E][socket_timer.c:14] SocketTimerSetSocket(): Adds socket 57 to socket timers
19:47:45.425 -> [E][socket_timer.c:21] SocketTimerSetLastUpdate(): Sets time stamp for socket 57
19:47:45.425 -> [E][socket_timer.c:21] SocketTimerSetLastUpdate(): Sets time stamp for socket 57
19:47:45.425 -> [E][generic_networkhandler.c:783] HandleDataOnTcpSocket(): TCP reply sent:
19:47:45.425 -> [E][socket_timer.c:21] SocketTimerSetLastUpdate(): Sets time stamp for socket 57
19:47:45.425 -> [E][generic_networkhandler.c:949] CheckAndHandleConsumingUdpSockets(): start
19:47:45.425 -> [E][generic_networkhandler.c:614] HandleDataOnTcpSocket(): Entering HandleDataOnTcpSocket for socket: 57
19:47:45.459 -> [E][generic_networkhandler.c:745] HandleDataOnTcpSocket(): Data received on tcp:
19:47:45.459 -> [E][encap.c:158] HandleReceivedExplictTcpData(): Handles data for TCP socket: 57
19:47:45.459 -> [E][encap.c:214] HandleReceivedExplictTcpData(): Send Request/Reply Data
19:47:45.459 -> [E][cpf.c:246] CreateCommonPacketFormatStructure(): Sockaddr type id: 8001
19:47:45.459 -> [E][cipmessagerouter.c:183] NotifyMessageRouter(): NotifyMessageRouter: routing unconnected message
19:47:45.459 -> [E][cipmessagerouter.c:219] NotifyMessageRouter(): NotifyMessageRouter: calling notify function of class 'connection manager'
19:47:45.529 -> [E][cipmessagerouter.c:137] GetCipInstance(): start instance_number:1
19:47:45.529 -> [E][cipcommon.c:88] NotifyClass(): notify: found instance 1
19:47:45.529 -> [E][cipcommon.c:99] NotifyClass(): notify: calling ForwardOpen service
19:47:45.529 -> [E][cipconnectionmanager.c:509] ForwardOpen(): start
19:47:45.529 -> [E][cipconnectionobject.c:224] ConnectionObjectSetState(): CIP_CONNECTION_OBJECT_STATE_NON_EXISTENT
19:47:45.529 -> [E][cipconnectionobject.c:157] ConnectionObjectInitializeFromMessage(): ForwardOpen: ConConnID 148697184, ProdConnID 148697185, ConnSerNo 36304
19:47:45.564 -> [E][cipconnectionmanager.c:553] ForwardOpen(): We have a Non-Null request
19:47:45.564 -> [E][cipconnectionmanager.c:562] ForwardOpen(): We have a Non-Matching request
19:47:45.564 -> [E][cipconnectionmanager.c:569] ForwardOpen(): end
19:47:45.564 -> [E][cipconnectionmanager.c:406] HandleNonNullNonMatchingForwardOpenRequest(): start
19:47:45.564 -> [E][cipconnectionobject.c:353] ConnectionObjectGetTransportClassTriggerProductionTrigger(): start
19:47:45.564 -> [E][cipconnectionmanager.c:1084] ParseConnectionPath(): start
19:47:45.564 -> [E][cipconnectionmanager.c:1088] ParseConnectionPath(): remaining_path : 4
19:47:45.599 -> [E][cipconnectionobject.c:591] ConnectionObjectSetProductionInhibitTime(): production_inhibit_time :256
19:47:45.599 -> [E][cipconnectionmanager.c:1116] ParseConnectionPath(): kSegmentTypeLogicalSegment
19:47:45.599 -> [E][cipconnectionobject.c:353] ConnectionObjectGetTransportClassTriggerProductionTrigger(): start
19:47:45.599 -> [E][cipepath.c:333] CipEpathGetLogicalValue(): start
19:47:45.599 -> [E][cipepath.c:336] CipEpathGetLogicalValue(): logical_format : 0
19:47:45.599 -> [E][cipconnectionmanager.c:1191] ParseConnectionPath(): class_id : 4
19:47:45.634 -> [E][cipconnectionmanager.c:1210] ParseConnectionPath(): classid 4 (assembly)
19:47:45.634 -> [E][cipconnectionmanager.c:1217] ParseConnectionPath(): remaining_path : 4
19:47:45.634 -> [E][cipconnectionmanager.c:1219] ParseConnectionPath(): remaining_path : 3
19:47:45.634 -> [E][cipepath.c:333] CipEpathGetLogicalValue(): start
19:47:45.634 -> [E][cipepath.c:336] CipEpathGetLogicalValue(): logical_format : 0
19:47:45.634 -> [E][cipconnectionmanager.c:1227] ParseConnectionPath(): Configuration instance id 1
19:47:45.634 -> [E][cipmessagerouter.c:137] GetCipInstance(): start instance_number:1
19:47:45.704 -> [E][cipconnectionmanager.c:1236] ParseConnectionPath(): remaining_path : 3
19:47:45.704 -> [E][cipconnectionmanager.c:1238] ParseConnectionPath(): remaining_path : 2
19:47:45.704 -> [E][cipconnectionobject.c:388] ConnectionObjectGetTransportClassTriggerTransportClass(): start
19:47:45.704 -> [E][cipconnectionobject.c:415] ConnectionObjectGetTransportClassTriggerTransportClass(): transport_class_trigger : 1
19:47:45.704 -> [E][cipconnectionmanager.c:1319] ParseConnectionPath(): assembly: type bidirectional
19:47:45.704 -> [E][cipepath.c:333] CipEpathGetLogicalValue(): start
19:47:45.739 -> [E][cipepath.c:336] CipEpathGetLogicalValue(): logical_format : 0
19:47:45.739 -> [E][cipconnectionmanager.c:1343] ParseConnectionPath(): connection point 150
19:47:45.739 -> [E][cipmessagerouter.c:137] GetCipInstance(): start instance_number:150
19:47:45.739 -> [E][cipepath.c:333] CipEpathGetLogicalValue(): start
19:47:45.739 -> [E][cipepath.c:336] CipEpathGetLogicalValue(): logical_format : 0
19:47:45.739 -> [E][cipconnectionmanager.c:1343] ParseConnectionPath(): connection point 100
19:47:45.739 -> [E][cipmessagerouter.c:137] GetCipInstance(): start instance_number:100
19:47:45.774 -> [E][cipconnectionmanager.c:1424] ParseConnectionPath(): Resulting PIT value: 256
19:47:45.774 -> [E][cipconnectionmanager.c:1427] ParseConnectionPath(): end
19:47:45.774 -> [E][cipconnectionmanager.c:431] HandleNonNullNonMatchingForwardOpenRequest(): ParseConnectionPath connection_status:0
19:47:45.774 -> [E][cipconnectionmanager.c:1517] GetConnectionManagementEntry(): class_id 4
19:47:45.774 -> [E][cipconnectionmanager.c:1521] GetConnectionManagementEntry(): i: 0 -> g_connection_management_list[i].class_id : 2
19:47:45.774 -> [E][cipconnectionmanager.c:1521] GetConnectionManagementEntry(): i: 1 -> g_connection_management_list[i].class_id : 4
[... 266 lines cut ...]
19:47:54.791 -> [E][cipconnectionmanager.c:776] AssembleForwardOpenResponse(): message_router_response->reply_service : 212
19:47:54.791 -> [E][cipconnectionmanager.c:777] AssembleForwardOpenResponse(): message_router_response->general_status : 1
19:47:54.791 -> [E][cipconnectionmanager.c:794] AssembleForwardOpenResponse(): AssembleForwardOpenResponse: sending error response
19:47:54.791 -> [E][cipconnectionobject.c:224] ConnectionObjectSetState(): CIP_CONNECTION_OBJECT_STATE_NON_EXISTENT
19:47:54.825 -> [E][cipconnectionmanager.c:857] AssembleForwardOpenResponse(): end
19:47:54.825 -> [E][cipmessagerouter.c:238] NotifyMessageRouter(): notifyMR: notify function of class 'connection manager' returned a reply
19:47:54.825 -> [E][socket_timer.c:21] SocketTimerSetLastUpdate(): Sets time stamp for socket 60
19:47:54.825 -> [E][generic_networkhandler.c:783] HandleDataOnTcpSocket(): TCP reply sent:
19:47:54.825 -> [E][socket_timer.c:21] SocketTimerSetLastUpdate(): Sets time stamp for socket 60
19:47:57.637 -> [E][generic_networkhandler.c:949] CheckAndHandleConsumingUdpSockets(): start
19:47:57.637 -> [E][generic_networkhandler.c:614] HandleDataOnTcpSocket(): Entering HandleDataOnTcpSocket for socket: 60
19:47:57.637 -> [E][opener_error.c:28] GetErrorMessage(): start error_number :128
19:47:57.637 -> [E][generic_networkhandler.c:641] HandleDataOnTcpSocket(): networkhandler: socket: 60 - connection closed by client: 128 - Socket is not connected
19:47:57.637 -> [E][encap.c:835] RemoveSession(): encap.c: Removing session
19:47:57.637 -> [E][cipconnectionobject.c:388] ConnectionObjectGetTransportClassTriggerTransportClass(): start
[... cut ...]
19:47:57.671 -> [E][generic_networkhandler.c:1014] CloseSocket(): networkhandler: closing socket 60
19:47:57.671 -> [E][generic_networkhandler.c:1020] CloseSocket(): networkhandler: closing socket done 60
19:47:57.671 -> [E][encap.c:835] RemoveSession(): encap.c: Removing session
19:47:57.671 -> [E][encap.c:843] RemoveSession(): encap.c: Session removed

 when socket 57 close ,should we close socket 58 and socket 59,
then socket 60 arrived,
we should not get kConnectionManagerExtendedStatusCodeErrorOwnershipConflict .

CapXilinx, 2019-09-02:
You do not send a ForwardClose according to your trace. Timed-Out Class 0/1 connections are not automatically deleted (see CIP Vol. 1 Fig 3-4.17).
As the EO connection therefore still exists, you correctly get an Ownership Conflict

yanzixiang, 2019-09-02:
> You do not send a ForwardClose according to your trace. Timed-Out Class 0/1 connections are not automatically deleted (see CIP Vol. 1 Fig 3-4.17).
> As the EO connection therefore still exists, you correctly get an Ownership Conflict

So If the scanner don't send ForwardClose,
It can NEVER register a session again ?

It seems that socket 59 is created for kUdpCommuncationDirectionProducing,
but why it did not send out any packet ?
which method should I trace for ?

CapXilinx, 2019-09-02:
You mean a new exclusive owner connection, (encapsulation) session is something different. And yes, as long as it not closed or the device is reset, it cannot create a new connection. But it is possible to reclaim the connection from the original creator. Not completely sure how this is checked, but I think the connection triad has to be the same.

I don't know why it is not sending. Perhaps you removed the check for capabilites at the start of the main, but you are not allowed to send raw data packet? Or your LWIP stack is not capable of creating a RAW socket?

yanzixiang, 2019-09-02:
> I don't know why it is not sending. Perhaps you removed the check for capabilites at the start of the main, but you are not allowed to send raw data packet? Or your LWIP stack is not capable of creating a RAW socket?

@the check for capabilites at the start of the main
where ?

@you are not allowed to send raw data packet
It is freeRTOS,I think there is NO permission problem.

@your LWIP stack is not capable of creating a RAW socket
which method should send out the packet ? I think I should dig into it.

CapXilinx, 2019-09-02:
`int main(int argc,
         char *arg[]) {

  cap_t capabilities;
  cap_value_t capabilies_list[1];

  capabilities = cap_get_proc();
  if (NULL == capabilities) {
    printf("Could not get capabilities\n");
    exit(0);
  }

  capabilies_list[0] = CAP_NET_RAW;
  if (-1
      == cap_set_flag(capabilities, CAP_EFFECTIVE, 1, capabilies_list,
                      CAP_SET) ) {
    cap_free(capabilities);
    printf("Could not set CAP_NET_RAW capability\n");
    exit(0);
  }`

This part in the POSIX main.c, starting from line 50.

I am no expert on FreeRTOS, but on Linux like OSes, you need the NET_RAW capability to create sockets with protocol type RAW.

```
EipStatus SendUdpData(struct sockaddr_in *address,
                      int socket_handle,
                      EipUint8 *data,
                      EipUint16 data_length)
```
is sending the data.

yanzixiang, 2019-09-02:
There is NO 
# sys/capability.h 
file in FreeeRTOS.
Oh........

CapXilinx, 2019-09-02:
Maybe FreeRTOS has not implemented capabilites right now. In that case try to run OpENer as a priviledged process

closed 2019-09-02 by yanzixiang