EIPStackGroup/OpENer #257: Add NV mechanism

CapXilinx, 2019-09-02:
Several objects have attributes which have to be persisted after a reset or reboot of the device. OpENer therefore needs some hooks and mechanisms to handle the persistance of attribute values to NV ram or disk (SD, SDD, HD, or whatever else)

stm-at-esd, 2019-09-11:
Hi,

I would like to tell you that I'm on the way to have a proof of concept implementation ready for the
non volatile storage of the QoS data. The QoS object itself is very simple and therefore needs only small changes. This avoids to clutter the view on the changes in the OpENer core. 

Perhaps end of the week there will be some patches.

Best regards,
    Stefan

stm-at-esd, 2019-09-19:
Based on the pull request proposal #262 I had a look on the qos.c and qos.h source. I did already another comment there (but I don't find it again now ...).

For the NV data patch the new code needs access to the QoS object and other object's data. The way how to achieve this I tried to explain [here](https://groups.google.com/d/msg/eip-stack-group-opener-developers/oRmN9WvGljk/3yQ16QbKBgAJ) already. Therefore I provided the example for the TCP/IP object.

From a pure object oriented view all object data should be hidden in the "object".c module and only be accessed by getter and setter functions. The current implementation of the stack relies heavily on global variables. That's good to see in the generic_networkhandler.c where the code accesses stuff like 
> extern CipTcpIpNetworkInterfaceConfiguration interface_configuration_

The NV data store code needs also access to a lot of data attributes. But my way to go would be at the moment not the pure object oriented approach.

I would provide a typedef for the "ObjectName" object in the cip<ObjectName>.h header that constitutes the CIP object. Then in the cip<ObjectName>.c source the object would be defined as "g_<ObjectName>" global variable which in turn would be made public in the cip<ObjectName>.h header by "extern Cip<ObjectName> g_<ObjectName>" variable.

For objects with multiple but statically known instances the global variable would be an array of Cip<ObjectName>. This would be true for the EthernetLink Object with three instances in the DLR case. Then the access to the global object instance would be "g_<ObjectName>[instance-1]".

As a later extension (not needed yet) the CipInstance structure could gain a "void *instance_data" pointer to the object data structure belonging to the CipInstance of that object type.

I would do this kind of data aggregation into single "Cip<ObjectName>" structure types for each CIP object. 

[cipqos_proposal.zip](https://github.com/EIPStackGroup/OpENer/files/3632594/cipqos_proposal.zip)

In the archive there are two cipqos_chg.[ch] files that have the changes for the NV data patch based on the Issue259 branch head of today.  This would be the starting point for me for the NV data patch.
I have the NV data saving backend (application level) ready and a proof of concept example code for the OpENer application based on a simple file (that only for QoS yet).

I would even change the sources more in a way that makes it for me easier to read (not so much doubled "dscp" in the names). These are the source files cipqos_more_chg.[ch] added to this post. if you're OK with it I would start with these versions.

As soon as the Issue259 branch gets merged to master I will start with the NV data patch for the OpENer common parts.

CapXilinx, 2019-09-20:
I am totally fine with your changes, I even like the more_chg more. You are right, that I replicated dscp too often :)

From your comment I assue you are also ok with my changes to #262 ?

Best,
Martin

stm-at-esd, 2019-09-20:
Yes, I think the changes of #262 are ready to be merged into  the master branch.

At first I would then commit the changes from the qos_more_chg.[ch]. After that I will add my NV data addition to the stack / library part of OpENer.

At the moment I have for OpENer only for the QoS object a proof of concept level implementation to store the NV data in plain C files (using standard FILE* pointers). It will create in the working directory 
where OpENer runs a file as "./nvdata/qos.cfg". The idea is to add other files like "./nvdata/tcpip.cfg" and "./nvdata/ethlink.cfg" as needed for the other object types.

Therefore I have some files that are (nearly) plain C that will be working for WIN32, MINGW and POSIX. These files (conffile.[ch], nvqos. [ch]) would go into a new directory ./source/src/ports/nvdata/
and a common handler with a function NvLoad(void) would be in ./source/src/ports/nvdata.c.

I hope it is ok to add another directory ./source/src/ports/nvdata/ in the source tree.

Best regards,
    Stefan

CapXilinx, 2019-09-20:
Ok, thanks for the feedback!

Sure it is ok to add directories, as long as it serves the purpose of making things more orderly.
Looking forward to your NV patch!

Best regards,
Martin

stm-at-esd, 2019-09-27:
Hi,

before I send the big NV data patch I would like to provide all the cleanup and preparation patches I
have here till today.

The patches should apply clean on the master branch. The purpose of the patches should become
clear from the commit message.

The patch 0003 is an exception and should be handled different. The commit message is misleading because this patch was a revert on my development branch that I did to make the merge of the Issue259 branch easier. But at the end after the merge it only changed some comments and whitespace. Therefore after applying that patch you should amend the commit message to become 
expressive and clear for the master branch.

Here comes the patch archive:
[patches_190927.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3664077/patches_190927.tar.gz)

Best regards,
    Stefan

PS: The NV data stuff for QoS and OpENer is now tested and working here.

stm-at-esd, 2019-09-30:
Hi,

unfortunately I missed your changes of last week on the master branch. Therefore I need to rebase the patches I sent on Friday on your latest version.

Best regards,
    Stefan

CapXilinx, 2019-09-30:
Hi,
you can also tell me to which commit your patches apply. I can also rebase/merge them if you want.

Best regards,
Martin

stm-at-esd, 2019-09-30:
Hi,

the rebase is done. I attach here the patches as they would apply on the head of master as of this morning. The comment on the patch 0003 as mentioned in my message of Friday applies also to this sequence of patches.

Here comes the archive with thirteen patches:
[patches_190930.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3671222/patches_190930.tar.gz)

The patch archive of 2019-09-27 (aka. patches_190927.tar.gz) should be discarded.

Best regards,
    Stefan

CapXilinx, 2019-09-30:
Hi!

Thanks for the patch.  I am currently reviewing and testing it.

Best regards,
Martin

stm-at-esd, 2019-10-01:
Hi,

the real NV data patch is still missing. But it is ready for commit and rebase here. I think tomorrow afternoon I will upload it.

Best regards,
Stefan

CapXilinx, 2019-10-01:
Oh sorry. I had to fix some stuff and thought it done then. Reopened the issue

stm-at-esd, 2019-10-02:
Hi,

here comes the patch that adds the NV data support:
[patches_191002.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3681758/patches_191002.tar.gz)

Here is the quote of the commit message:

> This patch adds the capability to OpENer to save attributes of objects to
> non volatile storage that are required to be Non Volatile by the spec.
> 
> The basic idea is that the application can connect a PostSetCallback handler
> to an object class. That handler will be called by the stack in any
> SetAttributeSingle* routine for each attribute of the class that has
> kNvDataFunc set in its "attribute_flags". The handler itself has the choice
> how to save this attribute or all objects attributes at once.
> 
> This patch only connects the NV data of the QoS object to the appropriate
> callback handler in the sample_application. It has only a proof of concept
> code level implementation to save the data in a configuration file
> nvdata/qos.cfg located in the working directory. The NV data load and store
> code is located in source/src/ports/nvdata subdirectory and should work for
> a standard C environment with file support.
> 
> The concept of service callbacks for Set* and Get* services was extended
> and therefore these callbacks were added to the CipClass type:
>  - PreGetCallback()
>  - PostGetCallback()
>  - PreSetCallback()
>  - PostSetCallback()	(also used for the NV data storage)
> 
> As a first use of the callbacks PreGetCallback() and PostSetCallback() the
> Assembly class now connects the applications BeforeAssemblyDataSend() and
> AfterAssemblyDataReceived() to the GetAttributeSingle / SetAttributeSingle
> implementation. This way the class specific code for the Assembly class in
> GetAttributeSingle of cipcommon.c could be removed.
> 
> The Pre- and Post- callbacks can also be used by the application to install
> a locking mechanism for the explicit requests like it was proposed in issue
> #222.
> 

My next step will be to make the attributes 3, 5 and 6 of the TCP/IP settable from the network
and also prepare these attributes to be stored in NV storage. The setting of the TCP/IP 
interface configuration will need a device reset to take effect (attribute 2 Configuration
Capability Bit 6=1) and set Bit 5 of Status to signal this condition.

For the QoS object I save the data only in an unstructured file. For the OpENer in general it
might be a good idea to use an existing library to save the NV configuration in something like
an *.INI file (one per object class?). I have already collected a list of suitable libraries but
atm. I don't have access to it.

Best regards,
    Stefan

PS.: Because of tomorrow's holiday (in Germany), I won't be back at the office until Friday.

CapXilinx, 2019-10-03:
Hi!
Thanks again for the patch. For the serialization of the object data, I would like to propose the JSON format, as it is easily human-readable and modifiable.
There are some open-source libraries available (see http://www.json.org/) which are written in C, e.g. https://github.com/DaveGamble/cJSON which is written in ANSI C like OpENer and has a compatible license (MIT).

CapXilinx, 2019-10-03:
Hi!

I just tried to apply your patch, but is seems you commited some stuff in the soruce/src/ports/POSIX/CMakeLists.txt which is not in the master branch ( if(OpENer_TRACES_ZLOG)
 else(OpENer_TRACES_ZLOG))

CapXilinx, 2019-10-03:
I fixed the patch and applied it in https://github.com/EIPStackGroup/OpENer/pull/269

commit ec4e88dd (2019-10-01): Merge pull request #268 from EIPStackGroup/Issue_257

Closes #257
commit f58a0aa3 (2019-10-10): Merge pull request #269 from EIPStackGroup/Issue_257_Add_NV

Closes #257 adds NV mechanism for QoS object
closed 2019-10-10 by CapXilinx