EIPStackGroup/OpENer #270: Integrate a simple EIP Scanner

timoc, 2019-10-04:
There is an open scanner, but it was buried in another open project abandoned years ago. Someone else seems to have had the same idea about pulling it out to see how useful it is. 

so Here is a simple "scanner" implementation to integrate, or possibly use as reference:
https://github.com/nismoryco/tuxeip

CapXilinx, 2019-10-04:
Hi,

thanks for the hint, but unfortunately the applied GPLv2 license from the original tuxeip project is incompatible with OpENer's modified BSD license. So any development has to be done from scratch, in order to not violate licenses.

Best regards,
Martin

timoc, 2019-10-07:
What about this one - it's BSD.
https://github.com/ros-drivers/odva_ethernetip

timoc, 2019-10-07:
This one is 'Adapted BSD'
https://github.com/liftoff-sr/CIPster

azoitl, 2019-10-07:
CIPster is OpENer based.

CapXilinx, 2019-10-07:
As @azoitl said, CIPster is a OpENer based fork, which follows a dual licensing scheme. As far as I am aware, CIPster does not have any Scanner capabilities, like OpENer.

For odva_ethernetip, this could be an example for a Scanner implementation, but is uses a completely different API as OpENer.

If there is a need for a Scanner application and there is support, I would be willing to start writing one. From my perspective I would still write it mainly from scratch, using OpENer itself as a base library and libevent or something similar for the communication handling.

timoc, 2019-10-08:
The scanner is the other side of this protocol. Having one that works with OpENer would be the goal.

CapXilinx, 2019-10-08:
OpENer works with a variety of Scanners (Rockwell PLC, Omron, CoDeSys ,ENIP), so getting a working scanner is not the point. Getting an open-source'ish (same license would have to apply for me) would be the benefit of this additional project.

nefethael, 2020-04-14:
Hello all,

There is also https://github.com/nimbuscontrols/EIPScanner project, don't know if you already looked at it? 

Best regards,
Vincent

CapXilinx, 2020-04-14:
Hi.

Thanks for bringing this to my attention, I didn't know about this project.

Best regards,
Martin

closed 2021-11-02 by CapXilinx