EIPStackGroup/OpENer #272: opENer can't answer my Forward_Open request

Moms91, 2019-10-17:
Hi Everyone, 

I have an issue who has been already mentionned (and closed by now) but I didn't really understand the solution.

When I send a Forward_Open to OpENer, then I get the error:
Invalid segment in connection path (0x0315)

I red that opENer is only compatible with packed Epath.
Please have a look at my wireshark packets and let me know what's wrong.
I suspect that given I send configuration data (simple segment data) in the connection path, opENer can't parse it, is it ?
However I can't tell my EtherNet tool not to send this configuration data once I select Exclusive Owner connection.
[EIP_opener_2.zip](https://github.com/EIPStackGroup/OpENer/files/3738075/EIP_opener_2.zip)

CapXilinx, 2019-10-17:
Hi,

OpENer should be able to parse the data segment, so this looks like a bug at first glance. Could you upload a trace log of OpENer for this? I will try to reproduce your forward open request with my setup in the meantime.

Best regards,
Martin

Moms91, 2019-10-18:
Hi Martin,

I'm a beginner so I'm afraid not to know how to retrieve the log. I sent you my whole rep so that you can find out. Thanks.
[OpENer-master.zip](https://github.com/EIPStackGroup/OpENer/files/3742604/OpENer-master.zip)

stm-at-esd, 2019-10-18:
Hi,

I compared the wireshark log you sent with a log of a working Forward_Open that I executed with the Ethernet/IP Explorer. Basically both tools create the same Forward_open request. But they differ in the Transport Type Trigger:
- The working Ethernet/IP Explorer (and another tool too) set a Transport Type Trigger of 0x01 i. e. signaling a client direction.
- The Mitsubishi (SPS?) sets a Transport Type Trigger of 0x81 i. e. signaling a server direction.

I think this is the culprit. But I'm not quite sure what that means for the reaction of OpENer. i. e. if the reaction of OpENer is correct or not.

See attached two screenshots of that decode the Forward_open request:
This is the working example
[image]
This is your request that was rejected by OpENer
[image]

Btw: The Connection Path stuff is equal for both requests.

Best regards,
    Stefan

Moms91, 2019-10-18:
Hi Stefan,

I modified the EDS file so to have the Direction as yours, but still the same error from opENer...

BR,
Moms

CapXilinx, 2019-10-25:
In your initial report you mentioned another issue similar issue. Did you mean Issue #225 ?
Because you do not have the same problem. Here the problem was, that the config path was different from it is configured in OpENer, but no config data was sent.

Your case is, that your paths should be correct and you are transporting the data segment payload.

I have to dig deeper into this, but right now I do not know what's wrong with your request

Best regards,
Martin

CapXilinx, 2019-10-25:
Ok, first finding. OpENer is in fact returning for your inital request the wrong error code. As it does not act as a server is should have returned 0x011E

CapXilinx, 2019-10-25:
So I did now almost the same Forward Open as you with the ENIPExplorer and OpENer accepts the Forward Open. So from what I see in the connection path settings, your request is correct and the same as my working one.
What's different thou and rather strange, your PLC assigns to both Network Connection IDs the value 0. I'm not sure if this is valid behavior.
Another point what's different to mine is the tick time (10 vs 5), but I don't know if this is really relevant.

From this point I would guess there is something before the connection path, which causes a hiccup in OpENer. I will try to investigate further, but perhaps you could also tryout ENIP Explorer, if this works with your setup.

Moms91, 2019-10-28:
Thanks for your reply.
I'll try with ENIP Explorer and check it out. Could you please send me your working F_Open frame.
Thanks.

BR,
Moms.

CapXilinx, 2019-10-28:
Hi Moms,

please find attached my Wireshark capture.

Best regards,
Martin

[OpENer_FO_with_Data_ENIP_filtered.zip](https://github.com/EIPStackGroup/OpENer/files/3778639/OpENer_FO_with_Data_ENIP_filtered.zip)

CapXilinx, 2020-01-15:
As there has been no activity for more than two months and no bug could be identified, I close this issue.

Please note there have been no such issue at the last PlugFest, so I guess OpENer is performing correct.

closed 2020-01-15 by CapXilinx