EIPStackGroup/OpENer #273: Ethernet Link object now supports multiple instances in preparation for DLR support

stm-at-esd, 2019-10-18:
Hi,

I changed the Ethernet Link object to support multiple instances which is needed for a DLR capable
device.

The patches I provide go on top of the patches provided with issue #271. This is the patch archive:
[patches_191018.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3745175/patches_191018.tar.gz)

Refer to the patch commit messages for further info.

The rework of the networkconfig.c module is still incomplete even for the POSIX variant. WIN32 and MINGW are untouched so far.

Best regards,
    Stefan

stm-at-esd, 2019-10-22:
Hi,

I had to amend my last patch of the uploaded patch archive. Therefore here comes the updated one:
[patches_191018b.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3757275/patches_191018b.tar.gz)

Best regards,
    Stefan

stm-at-esd, 2019-10-29:
Hi,

I rebased my current development branch to the master branch of yesterday (git hash 25b8c3faacace5fc2ba1968a2a1f41f942630e91). But I found out that during merge of cipstring.c 
an error crept in.

I provide here two patches:
[patches_191029.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3783204/patches_191029.tar.gz)

- 0001: This patch fixes the bug.
- 0002: This patch removes tab characters from the source code and makes the function prototypes 
  consistent to your changes.

These patches can be applied to current master and won't disturb my upcoming patch to add support for the Interface Counters and Media Counters attributes of the Ethernet Link object.

But I still have a question. You reformatted the cipstring.c source code for coding style reasons. During this you moved the include line for the "cipstring.h" from being the absolute first included header to below the included system headers (which are in <> brackets). But I had written the
include line for "cipstring.h" as first include by intent because the Google coding style you refer to
in your coding style recommendation requires it to do so.

How should it be done with the include of the own header of a C source module in the future?

After these patches applied, this issue could be closed then. I think I will open another issue to handle the new support for the counter attributes of the Ethernet Link object.

Best regards,
    Stefan

CapXilinx, 2019-10-29:
Oh sorry, it seems that I just messed up, you are absolutely correct. I thought libs are to be moved to top.
Sorry for causing you extra effort.

Best regards,
Martin

stm-at-esd, 2019-10-30:
Hi,

I will adjust the order of the header inclusion in cipstring.c then. The resulting order will then be:
- The header of the current C module i. e. `#include "mymodule.h`
- The headers installed in system directories like stdlib.h and so on i. e. `#include <stdlib.h>`
- The headers of the project without the own module header i. e. `#include "othermodules.h"` and so on

I have another question. Where should I put my new "cipdlr.c" and "cipdlr.h" source files of the DLR object implementation? I would prefer _source/src/cip/cipdlr.[ch]_. Or should it go into the optional
objects directory _source/src/cip_objects/cipdlr.[ch]_?

Best regards,
    Stefan

CapXilinx, 2019-11-06:
Hi,

sorry for the delay. I would also think it should go into the non-optional src/cip folder, as it will be probably tightly integrated into the main stack.

For the header files, thanks and sorry for the extra work.

Best regards,
Martin

CapXilinx, 2020-01-16:
Added with commit ef68f1502d1399d91cb8d3a5fe9ca3850b47e0be

Thank you Stefan!

closed 2020-01-16 by CapXilinx