EIPStackGroup/OpENer #274: How to build

smccoorp, 2019-10-23:
Hello, 
Can anyone please help me out for compiling this stack. i want to use this stack on "Raspbian OS". 
I have never done any stack compilation. i want to know its compilation process with steps and commands to be used for compilation.

CapXilinx, 2019-10-23:
Please take a look at the README file "Compile for Linux/POSIX". It is explained there

smccoorp, 2019-10-23:
Yes i have read that, actually i am new to this thing so i don,t know what command has to be for the process mentioned there. i will be thankful if you can tell the command which has to be write in terminal

CapXilinx, 2019-10-23:
The commands mentioned in the README are literally the commands you type into the terminal.

smccoorp, 2019-10-24:
in readme file it is memtioned ad:

**Compile for Linux/POSIX:**
Make sure all the needed tools are available (CMake, make, gcc, binutils)
Change to the /bin/posix
For a standard configuration invoke setup_posix.sh
Invoke the make command

Grant OpENer the right to use raw sockets via sudo setcap cap_net_raw+ep ./src/ports/POSIX/OpENer

Invoking OpENer:

./src/ports/POSIX/OpENer <interface_name>

e.g. ./src/ports/POSIX/OpENer eth1

OpENer also now has a real-time capable POSIX startup via the OpENer_RT option, which requires that the used kernel has the full preemptive RT patches applied and activated. If you want to use OpENer_RT, instead of step 2, the sudo setcap cap_net_raw,cap_ipc_lock,cap_sys_nice+ep ./src/ports/POSIX/OpENer has to be run to grant OpENEr CAP_SYS_NICE, CAP_IPC_LOCK, and the CAP_NET_RAW capabilities, needed for the RT mode

Shared library support has been added to CMakeLists file and is enabled by setting OPENER_BUILD_SHARED_LIBS=ON. It has only been tested under Linux/POSIX platform.

1. Change to the /bin/posix   **for this i used command "cd ../../bin/posix**
2. For a standard configuration invoke setup_posix.sh **what command has to be invoke setup_posix.sh**
3. invoke the make command **what command has to be invoke make**
4. Grant OpENer the right to use raw sockets via sudo setcap cap_net_raw+ep ./src/ports/POSIX/OpENer **i feel this is complete command mentioned here**
5. Invoking OpENer:
./src/ports/POSIX/OpENer <interface_name>
e.g. ./src/ports/POSIX/OpENer eth1 **what command has to be invoke OpENer**

Is there any extra steps apart of these 5 steps has to be done for compiling and i am having with what command has to be used. if anything is to be read first before doing steps please let me know, i will.

CapXilinx, 2019-10-24:
> Change to the /bin/posix for this i used command "cd ../../bin/posix
There is a bin/posix directory inside OpENer, use cd to change to it

> what command has to be invoke setup_posix.sh
In the bin/posix literally invoke the setup_posix shell script (`./setup_posix.sh`)

> Invoke the make command what command has to be invoke make
Literally `make`

> Grant OpENer the right to use raw sockets via sudo setcap cap_net_raw+ep ./src/ports/POSIX/OpENer i feel this is complete command mentioned here
Yes `sudo setcap cap_net_raw+ep ./src/ports/POSIX/OpENer` is the complete command, as the binaries are generated so the path should be the same

> e.g. ./src/ports/POSIX/OpENer eth1 what command has to be invoke OpENer
again literally `./src/ports/POSIX/OpENer` and the name of the ethernet interface

As said before, all these commands are already literally in the README, so please try them out, before asking the same question over and over again.

CapXilinx, 2019-11-06:
Closed due to inactivity

closed 2019-11-06 by CapXilinx