EIPStackGroup/OpENer #275: Off-Link(subnet) routing test for multicast connections.

desmith2, 2019-10-23:
Hi,

A request for an addition to correct a conformance test issue concerning the connection manager for "Off-Link(subnet) routing test for multicast connections".

In the case OpENer receives a multicast connection request over the segment between class B and class C, OpENer does not respond with: General Status 0x01 and Additional Status 0x0813 (NOT CONFIGURED FOR OFF-SUBNET MULTICAST) in the forward open reply. 

TTL is 1 (default) in this case. TTL > 1 would accept the connection.

Regards, 
Dennis

CapXilinx, 2019-10-24:
Hi,
thank you for your input. Can you also explain me your setup and (CT) configuration, so I can test it on-site here?

Best regards,
Martin

desmith2, 2019-10-24:
Thanks for the fast response.

So we have a motion controller using Ethercat for axis control and using EtherNet/IP on a private Class C subnet for IO, HMI and control.

Overall concern is for safety. So we want to reject multicast forward open requests on EtherNet/IP that are not part of our local subnet.
Some of the systems that these controllers are used on may have multiple routes, Class A/B or non-private multicast online.

I found the referenced error code located in cipconnectionmanager.h:

   kConnectionManagerExtendedStatusCodeNotConfiguredForOffSubnetMulticast = 0x0813,

I can't find any implementation.

So I'm assuming on a request for opening a forward multicast:

1) If the originator is in the same network accept the request.
2) If the originator is not in the same network:
    a) If the TCP/IP interface objects TTL == 1: 
    	Return General Status 0x01 and Additional Status 0x0813.
    b) If the TCP/IP interface object TTL > 1: 
    	Accept the request.
    	
Note: The TTL value here is the number of hops remaining the packet has along the path till destination. If the TTL reaches 1 then it wasn't found.

So to test this I would think setting the port that OpENer is on to a private Class C (192.168.100.x/24 as example).

Then expose multiple subnets, private/non-private and Class A/B on the same path. 
Multicast from these should be rejected.
    	
Regards,
Dennis

desmith2, 2019-10-27:
OOPS ... Sorry I missed the primary part of your question,

The test tool is CT-16. I'm not part of the group testing and have realized that they did not provide the STC file being used. I will get this to you ASAP.

Also sorry for all the detail in the previous reply. I was working through my own analysis.

Regard,
Dennis

CapXilinx, 2019-10-28:
Hi Dennis,

no problem, details are good. They also help me to understand the problem. Usually I only operate OpENer with TTL=1, so this bug is quite of interest for me.

Thanks for bringing this up!

Best regards,
Martin

desmith2, 2019-10-28:
Hi Martin,

I have attached the CT-16 configuration file that our testers are using.

[CK3M-CPU121.stc.zip](https://github.com/EIPStackGroup/OpENer/files/3779698/CK3M-CPU121.stc.zip)

Regards,
Dennis

CapXilinx, 2019-11-13:
Hi Dennis,

sorry for the late reply. Thanks for the config, I will dig into it as soon as I have time, but to be honest this could take some time, as I have to finish the development of the CIP File Object first.

Best regards,
Martin

stm-at-esd, 2020-02-20:
Hi,

this issue is fixed by patch number 0011 from the patch archive added to issue #285

CapXilinx, 2020-04-23:
Ok, I will close this issue as soon as I integrated the patches, so as soon as this closes (which is hopefully in the course of the next week) this should be fixed for you @desmith2 

Best regards and sorry for the delay
Martin

desmith2, 2020-04-25:
Thank you all.

CapXilinx, 2020-05-15:
Fixed in b6e6022

closed 2020-05-15 by CapXilinx