EIPStackGroup/OpENer #278: Complete object implementation for DLR object

stm-at-esd, 2019-11-15:
Hi,
 I have added now an implementation of the DLR object for a non-supervisor and non redundant gateway device. For the implemented features and attributes please have a look into the source/src/cip/cipdlr.[ch] files.

The implementation is only a front-end for a part of software that together with some dedicated hardware implements the DLR protocol on the EtherNet/IP network. It's more or less only a
means to read the internal status of the DLR protocol state machine from the EIP network.
That means without a hardware/software for the DLR protocol the Conformance Test sends some
DLR frames and gets no response and the DLR test fails.

I think that I now have included all required features for such a DLR device on the (OpENer) stack level. But the next weeks of testing will show that hopefully.

The list of patches to attach to this issue is relatively long. It is based on the master branch but the patches of issue #273 provided in the archive patches_191029.tar.gz still need to be applied.

Additionally to the changes for the DLR object there are two patches that need more explanation:
 * 0008: Fix-Add-more-checks-to-SetAttribute ...
   This patch does more checks on the SetAttribute service for the "Interface Configuration" 
   attribute. The Conformance Test did more exercises with the reconfiguration of the attribute
   than expected only from the spec.
 * 0010: Use-new-interface-to-read-configuration-from-network
    This patch changes the interface used to read the current configuration from the network 
    interface / system for the Ethernet Link object's MAC attribute and the TCP/IP object's interface
    configuration attribute. There is no hidden access to global object attributes any more. The read
    information is returned via parameters of the new functions. Also the functions have been 
    reorganized to retrieve the complete set of data for the "Interface configuration" attribute. Before
    the IP addresses were retrieved by one function and another part of the interface configuration
    the domain name was retrieved by another function. I believe the interface is clearer now and
    can be better used from the main() function.

The last patch is for POSIX only but I have a nearly ready to commit patch for the MINGW version, that was completed in my private time. The rework of the interface showed that the MINGW code was broken in some aspects. That made a complete rework of the MINGW/networkconfig.c unavoidable. Using the MINGW compiler in parallel also triggered some of the cleanup patches already in the current patch list.

Here comes the patch archive:
[patches_191115.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3853007/patches_191115.tar.gz)

Best regards,
    Stefan

PS.: Will you attend the EIP Plugfest in 's Hertogenbosch at the beginning of December? I will be there with a colleague.

CapXilinx, 2019-11-18:
Hi Stefan,

oh sorry, I thought you wanted to amend the contribution to fix my error. I will add the other fix asap, but currently I am sick, so it will probably take some time.

Thanks also for this contribution, which is, at least in my opinion, a great contribution to the community, as everyone goes for DLR devices.

Yes, I plan to attend there. I have still to book my flight and hotel, but we will see there in person!

Best regards,
Martin

stm-at-esd, 2019-11-19:
Hi Martin,

I'm sorry to hear about your sickness. I wish you a fast recovery.

For now I want to provide a patch to use the new interface to read the configuration from the network interface but now it is only the MINGW version needed.

As an addition the new functions are able to select the right network interface either by an interface index (as it was done before) or by the interface alias name that is shown in the Windows GUI. As
a side effect of using some newer WIN32 API functions (and members of the address table) the
code won't run under Windows XP any more. It needs at least a Windows Vista. I hope that this is
not a harsh restriction nowadays.

I have been testing the MINGW version and its working. Because I don't have a Microsoft Compiler at hand I won't provide a WIN32 version of these functions. But for someone having a Microsoft Compiler available, it should be only a case of Copy and Paste and then fixing some warnings in the code, I think.

I believe that the implementation for the DLR stuff and the required attributes is now complete.
But the next two weeks of testing will show.

Best regards,
    Stefan
[patches_191119.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3865164/patches_191119.tar.gz)

CapXilinx, 2020-02-03:
Merged and updated for MSVC in 6bd0f81c0d7c423d2c3c2464980039dd76941220

closed 2020-02-03 by CapXilinx