EIPStackGroup/OpENer #279: Bug: OpENer rejects the Listen Only Connection even if the Exclusive Owner Connection is open

stm-at-esd, 2019-11-26:
Hi,

I today stumbled over a behavior that is a bug, I believe.

For testing purposes I opened the Exclusive Owner Connection (Conf: 151, O->T: 150, T->O: 100). I see the connection data in Wireshark. Then I tried to establish also the Listen Only connection (Conf: 151 with zero configuration size, O->T: 153, T->O: 100). There are two cases to distinguish:

a) The Exclusive Owner Connection is opened as Multicast-Connection. The establishment of the Listen Only is successful (either as Multicast or as Point-to-Point Connection).

b) The Exclusive Owner Connection is opened as Point-to-Point Connection. The establishment of the Listen Only Connection is rejected by OpENer with the error codes (general: 0x01, extended: 0x0119).

The Ethernet/IP specification's explanation for that error code is:

> NON-LISTEN ONLY CONNECTION NOT OPENED
> Connection request fails since there are no non-Listen Only connections with
> the same T->O application path currently open. Refer to Section 3-6 for a
> description of application connection types
> 

Judging from the table 3-6.2 in Vol. 1 Section 3-6.3 I would say that the establishment of a new Listen Only connection should succeed for both cases (Exclusive Owner Connection in Multicast or Point-to-Point mode).

I think the reason for this failure is that OpENer in appcontype.c around line 293 only checks for existing Multicast connections with GetExistingProducerMulticastConnection() where it additionally should look for Unicast Connections with the same T->O data path.

Would you please comment on this.

Thanks,
    Stefan

CapXilinx, 2019-11-26:
Hi Stefan,

I would say you are correct. I also thought I encountered this at the last PlugFest and fixed it already once... Perhaps I wrongly merged it back.

Your proposed fix should do the trick.

Thanks for bringing this up!
Best regards,
Martin

stm-at-esd, 2019-11-27:
Hi Martin,

I think I could fix that bug. Patches are attached.
* Patch 0001) Fixes only a typo in a comment.
* Patch 0002) The real patch for the bug.

[patches_191127.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3898499/patches_191127.tar.gz)

Best regards,
    Stefan

CapXilinx, 2020-02-03:
Merged in f2b8fd90a43f82e5c2ad1dcd9bc811997d5df0c1

closed 2020-02-03 by CapXilinx