EIPStackGroup/OpENer #280: OpENer does not reject Connections with too small RPI

stm-at-esd, 2019-11-26:
Hi,

I have been doing some more playing around with connections. My OpENer is compiled with "kOpenerTimerTickInMilliSeconds = 10" for a 10ms timer tick. For load testing purposes I opened
the Exclusive Owner connection with the O->T and T->O RPI both set to 2ms in cyclic mode.

OpENer accepts this connection request and opens the connection. My originator sends the O->T packets in a 2ms cycle. OpENer only sends the T->O packets in a 10ms cycle. After rebuilding
OpENer with 5ms kOpenerTimerTickInMilliSeconds the T->O packets followed with 5ms distance
to each other.

After all I was not very astonished because looking at NetworkHandlerProcessOnce() the 
ManageConnections() function is only called all kOpenerTimerTickInMilliSeconds ms.

But what I would expect (also from my experience from DeviceNet) is that OpENer would reject all connections with a RPI below its timer tick resolution.

I think this behavior can be derived from Vol. 1 Section "3-5.4.1.2 Requested and Actual Packet Intervals". There the RPI (Requested Packet Interval) and API (Actual Packet Interval) are defined.

For the API some citations from that section:
> The actual packet interval (API) shall be the maximum time between packets in microseconds.

> The API shall equal the RPI within the device’s timer resolution limits, ...

> If the API is not equal to the RPI it shall be faster than the RPI unless the resulting API would 
> be 0, in which case the RPI can’t be supported and an error shall be returned, ...

For the API I understand this as follows:
- The API is the real time distance between packets (at least for cyclic connections).
- The API shall be rounded (towards zero) from the RPI taking the device's timer resolution limits into account. For OpENer this would be the kOpenerTimerTickInMilliSeconds.
- If the RPI is below the device's timer resolution (i. e. kOpenerTimerTickInMilliSeconds) then the resulting API would be 0 (zero) and therefore the matching error should be returned.

Therefore I would expect OpENer to return the appropriate error codes shown in that section and reject this connection.

For now I may be able to work around this issue by setting kOpenerTimerTickInMilliSeconds to 1 and hoping that nobody tries an RPI below 1ms and on the other hand I don't overload my device.

Can you please comment on this.

Thanks,
    Stefan

stm-at-esd, 2019-11-26:
Btw. I forgot to mention that OpENer returns in the response the T->O API value 2000 (us) which is the same value as requested in the forward open (was 2000 (us)). Despite of the fact OpENer internally knows that it can't fullfill this API promise with the 10ms kOpenerTimerTickInMilliSeconds value.

CapXilinx, 2020-02-03:
Merged and fixed for Windows in 6bd0f81c0d7c423d2c3c2464980039dd76941220

closed 2020-02-03 by CapXilinx