EIPStackGroup/OpENer #282: PlugFest 2019 fallout: Changes for Ethernet Link object to complete DLR support

stm-at-esd, 2019-12-18:
Hi Martin,

during the PlugFest it showed up that a DLR device needs to support at least Autonegotiation AND
forced 100MBit FullDuplex ethernet links.

To accommodate that I developed a patch that implements that Ethernet Link object's attribute 6 becomes settable and it also will be written to the NV data storage. So far the NV data storage
back-end is not included the same way as it is not with the other patches that need NV data storage.

These three patches are provided and can be applied on top of the patches released for issue #279:
 - 0001: Fix some file descriptor leakage for the POSIX port
 - 0002: Add support for the Interface Control attribute to become settable
 - 0003: Update Ethernet Link object's Doxygen comment

[patches_191218.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3979780/patches_191218.tar.gz)

Best regards,
    Stefan

CapXilinx, 2020-01-14:
Hi,

I am back from my little hiatus. I will start now adding the missing changes in the course of this week.
Thanks for the patches!

Best regards,
Martin

stm-at-esd, 2020-01-14:
Hi,

I would like to take the opportunity to add here another two patches for some stuff I found last week.
These patches fit nicely with the patches already listed here in the bug report.

 * 0001: Fix - CheckIoConnectionEvent() was called with wrong parameter order.
    Found this when I tried to implement the LED control for the Network LED based on the
    number of operating and timed-out connections.
 * 0002: Cleanup - Remove duplicate initialization of ENIPMessages
   Found this during debugging.

Find more explanation on the patches in their commit messages.

[patches_200114.tar.gz](https://github.com/EIPStackGroup/OpENer/files/4060603/patches_200114.tar.gz)

Best regards,
    Stefan

stm-at-esd, 2020-01-17:
Hi,

I'm doing the last steps with my device hopefully being complete in the next week. During the implementation of the control for the Module Status LED I found some extensions to the accessors
for the Identity object's "status" member helpful.

I' m providing these changes as a patch here.
[patches_200117.tar.gz](https://github.com/EIPStackGroup/OpENer/files/4079010/patches_200117.tar.gz)

Best regards,
    Stefan

CapXilinx, 2020-02-03:
Hi Stefan,

thanks for all the patches. I am a bit behind my schedule, as I had some strange error in intermediate versions. I will try to add all your provided patches fill my problems.

Best regards,
Martin

CapXilinx, 2020-02-03:
Merged in d8ac347bc3c2f21658c7bb2feacbd0856f234cb3

closed 2020-02-03 by CapXilinx