EIPStackGroup/OpENer #285: Collection of patches still needed to pass conformance testing

stm-at-esd, 2020-02-07:
Hi,

I've been recently at the conformance test lab and at last passed the conformance test.

During the preparation for that event some issues showed up and I will provide the patches for these issues in the near future. There are around 10 patches pending. One of them will fix the bug #275 .

I hope I can provide the patches in the next week. I'll need to have a look if I rebase my patches first on the current master branch or if I provide them as is.

Best regards,
    Stefan

stm-at-esd, 2020-02-18:
Hi,

here comes the first patch of the pending patches. It had been prepared some time ago but I didn't
manage to upload it till now. It should apply to the head of the master branch without problems.

 * 0001: If the Ethernet Link Interface Control attribute has been configured as an
existing attribute (OPENER_ETHLINK_IFACE_CTRL_ENABLE) the default value
used for initialization was wrong.

[patches_200123.tar.gz](https://github.com/EIPStackGroup/OpENer/files/4221287/patches_200123.tar.gz)

More patches to come tomorrow.

Best regards,
    Stefan

stm-at-esd, 2020-02-20:
Hi,

now at last I have prepared the patches that were needed to pass conformance testing with our device. Conformance Test was done with CT16. I rebased and reordered the patches on the master branch commit fa13addae5bf3a6c4420231ace6a3f9dc7d52a30

There are now 12 patches ready for inclusion. After the rebase the patch 0001 here is a duplicate of the patch from patches_200112.tar.gz. The older patch should be skipped therefore.

Description of the patches:

* 0001: Fixed default values for Ethernet Link Interface Control (attribute number 6)
* 0002: Fixed define OPENER_ETHLINK_IFACE_CTRL_ENABLE missing for MINGW and WIN32
  This patch extends your patch with commit id 224fa11c6d269b5f0ab5441e80e262226cc87d4c
* 0003: Cleanup: Silence some unused variable compiler warnings
* 0004: Cleanup: Unify the use of the OPENER_ASSERT() macro and fixed some invalid uses
  The standard C assert() macro is defined without a trailing semicolon. Some of the OPENER_ASSERT() macro definitions had a trailing semicolon and some not.

  To make all variants of the macro definition interchangeable the definitions
in all opener_user_conf.h have been heavily reworked.

  Now all OPENER_ASSERT() macro definitions lack a trailing semicolon. As a
consequence a semicolon had to be added on numerous uses of that macro in
the source.

  At some points OPENER_ASSERT() was called with a literal character string
as argument. This failed the intended use completely and was replaced by
an OPENER_TRACE_ERR() call with the literal string followed by an
OPENER_ASSERT(false).
* 0005: Cleanup: Renamed enum member to make its name more expressive
* 0006: Fix: Return kCipErrorAttributeNotSupported for unsupported Ethernet Link object attributes
* 0007: Fix: **Work around** possible race condition when detecting gateway on POSIX
  The function GetGatewayFromRoute() evaluates /proc/net/route to detect the
gateway. If the interface is configured by DHCP the DHCP client first sets
the IP address and then runs the scripts to setup the routing information.

  The main program waits for the IP address to appear on the network interface
and immediately calls IfaceGetConfiguration() to get all needed information.
But it appears to be possible that GetGatewayFromRoute() is called before
the script setting up the routing information finishes.

  This **workaround** adds a 300ms delay before calling GetGatewayFromRoute() to
avoid that issue.

  **WARNING**: Later testing showed that there may be circumstances that 300ms is **NOT** enough.

  **NOTICE**: At last we abandoned this **workaround** because it still contains a race condition which is only covered by the additional waiting time.
  In our setup the DHCP client calls the setup script _/etc/udhcpc.d/50default_ to setup the interface
  inclusive routing information etc. We added an additional script _/etc/udhcpc.d/55etip_ being called
**after** _50default_. This script creates a file _/var/tmp/etip_resolv.conf_ in the "ramdisk" on the "bound" event and deletes it on the "release" event. Now _GetDnsInfoFromResolvConf()_ polls for the appearance of that file which automatically vanishes during a reboot/power cycle (ramdisk!).
Because this solution is very system dependent we don't provide a patch.
* 0008: Fix possible inconsistency between TCP/IP object attributes 3 and 5
* 0009: Fix possible inconsistency between TCP/IP object attributes 3 and 5; option a)

  The patches 0008 & 0009 both address the same problem but implement solution option b) (patch 0008 only applied) or implement solution option a) (both patches applied in order). We chose option b) for our device. But the Conformance Test Tool detects is solution as an error (flaw in the tool). But according to the conformance test lab this option is perfectly valid. To avoid confusion for EtherNet/IP novices implementing option a) in OpENer may be better. For detailed description see the patch comments and refer to PUB00049, Rev. 1.1, "EtherNet/IP Interoperability Conformance
Test Specification" in the sections 4.2.1.1.3 & 4.2.1.1.4.

* 0010: Fix TCP/IP object's attribute 3 being settable to an unsupported IP configuration method
* 0011: Fixes to pass the Off-Link Routing Test Procedure of Conformance Test
  This is a fix for the bug described in bug report #275. Ironically I missed that bug report in my
  preparation for the conformance test. The failure was found during the conformance manual
  testing and then the code to fix the issue was added.
* 0012: Fix: Must have "active" IO connections to reject setting of TCP/IP object's attribute number 5
  The conformance test lab criticized that even a broken IO connection in the timed-out state prevents the IP configuration from being set. Only "active" connections should prevent that according to the specification. Now we no longer consider IO connections in the timed-out state as active.

Here is the patch archive:
[patches_200220.tar.gz](https://github.com/EIPStackGroup/OpENer/files/4231126/patches_200220.tar.gz)

Best regards,
    Stefan

stm-at-esd, 2020-02-20:
Hi,

another quick patch for a bug that came to my attention by https://groups.google.com/d/msg/eip-stack-group-opener-developers/kRfU6WSwmPw/ZQkWqFGmAwAJ

Created this small patch to fix that issue:
[patches_200220-2.tar.gz](https://github.com/EIPStackGroup/OpENer/files/4231728/patches_200220-2.tar.gz)

The bogus error message can be triggered by using the Conformance Test Tool and selecting the "Connection Manager" test and the subtest "Forward Close". Then a lot of the bogus messages are triggered without the fix.

Best regards,
    Stefan

CapXilinx, 2020-04-23:
Hi Stefan,

thanks for all your work here. Your fixes really improved OpENer. I will try to integrate your changes ASAP, unfortunately I was quite busy in the last months with other projects, such as first things for CIP Safety. 

Best regards,
Martin

commit b6e6022a (2020-05-15): Merge pull request #292 from EIPStackGroup/Issue_285

Fixes #285 #292 #275
closed 2020-05-15 by CapXilinx