EIPStackGroup/OpENer #287: Unused function

jvalenzuela, 2020-03-08:
ConnectionObjectGetInstanceTypeForAttribute() in cipconnectionobject.c has some signed/unsigned issues with its return value, however, that function doesn't appear to be used anywhere. Is it ok to simply remove this function instead of working on fixing the return value?

Applies as of fa13add.

CapXilinx, 2020-05-12:
Hi,

I have to look on this first. Either I added it and didn't had time to bring it into use, it was for a future update, or  it is really not needed.

CapXilinx, 2020-09-02:
"GCC now defaults to -fno-common. As a result, global variable accesses are more efficient on various targets. In C, global variables with multiple tentative definitions now result in linker errors. With -fcommon such definitions are silently merged during linking."

Perhaps this is the issue