EIPStackGroup/OpENer #289: Use localhost as the network adaptor

iceblu3710, 2020-05-07:
It would be awesome if I could use localhost as the adaptor as I want to use OpENer as an interface to a desktop c# app. This would all be running with a Codesys softPLC on the same machine. Currently, if I use the same adaptor the communications just conflict one another.

azoitl, 2020-05-07:
The problem here is not localhost the problem is the port address. AFAIK the EIP spec demands that an EtherNet/IP adapter listens to several dedicated port addresses (e.g., 0xAF12). As this is mandatory these ports are currently hardcoded in OpENer's code. I don't think that changing the port addresses will result in a valid EtherNet/IP device, and therefore I don't see the possibility to run two adapter on the same machine.

CapXilinx, 2020-05-07:
Yes, as Alois says, the ENIP specification demands that traffic is transported via the same specified port, from both originator (PLC) and target (ENIP device). This is fundamental for how communication works and cannot be changed, as it would not be clear which application should receive the package. You need two machines (or VMs) for this to work.

If this was somehow possible to do, it would be probably to much effort to implement this for this quite unusual use-case, but if you want to try it and it can be done without compromising the main functionality, I am willing to add this to the repository.

iceblu3710, 2020-05-07:
Similarly there is no way to specify  new port on the ENIP Scanner so that
wouldn't help.

I think I can virtualize the network card into two virtual ones. One will
mirror the original settings and a new one will pull a new IP and have its
own Mac. Not as user friendly or all in one but it should work.

I wonder if there is a pure C# way to get a second IP without
virtualization.

azoitl, 2020-05-07:
Maybe a Docker container can help here.

iceblu3710, 2020-05-12:
Turns out you can multi-home a single interface with a static IP. This way I can run my softPLC on the "Ethernet 2:1" IP and OpENer on "Ethernet 2".

As a general bug report "Ethernet 2:1" is not recognized.

closed 2020-05-12 by iceblu3710

rednexela1941, 2023-05-30:
In case anyone runs into this on linux, the basic recipe is:
```bash
#!/usr/bin/env bash

# create two new network namespaces.
sudo ip netns add adapter_namespace
sudo ip netns add scanner_namespace

# connect with veth pair.
sudo ip link add veth1 type veth peer name veth2

# assign veth pair to namespaces.
sudo ip link set veth1 netns adapter_namespace
sudo ip link set veth2 netns scanner_namespace

# add ip addresses.
sudo ip netns exec adapter_namespace ip addr add 10.1.1.1/24 dev veth1
sudo ip netns exec scanner_namespace ip addr add 10.1.1.2/24 dev veth2
sudo ip netns exec adapter_namespace ip link set veth1 up
sudo ip netns exec scanner_namespace ip link set veth2 up

# OpENer depends on /proc/net/route and will look for the target interface.
sudo ip netns exec server_namespace ip route add default via 10.1.1.1 dev veth1
sudo ip netns exec client_namespace ip route add default via 10.1.1.2 dev veth2

# setup loopbacks.
sudo ip netns exec adapter_namespace ip link set lo up
sudo ip netns exec scanner_namespace ip link set lo up

# must support mutlicast.
sudo ip netns exec adapter_namespace route add -net 224.0.0.0 netmask 240.0.0.0 dev veth1
sudo ip netns exec scanner_namespace route add -net 224.0.0.0 netmask 240.0.0.0 dev veth2

: <<HERE
run programs using:
$ sudo ip netns exec adapter_namespace ./OpENer veth1
$ sudo ip netns exec scanner_namespace ./scanner
HERE
```