EIPStackGroup/OpENer #290: Win32 Build Issues

iceblu3710, 2020-05-12:
I am trying to build on Windows 10 and compilation always says 9 succeded and 1 skipped. The one skipped is `BUILD_ALL` but as it's listed last I assumed it has just a catch-all. It is de-selected in the solution configuration, I can check `build` but it does nothing. I also can not set the sample application as the `DEPLOY`

When running OpENer though I do not get any ENIP traffic on Wireshark. The ENIP session is registered but no `O->T` communications happen. When I build on Linux I do get proper ENIP traffic and the sample application is running. (I set outputs and they are mirrored on the inputs). I do not think the Win32 version is loading the sample applications code.

When I define `OPENER_TRACE_LEVEL` to see what is going on the Win32 version says `unresolved external symbol _LOG_TRACE` but Linux does not complain of this and I can get logging info.

I am using Windows 10 and Visual Studio 2017.

CapXilinx, 2020-05-12:
That are actually three different issues.

The first is thing with the skipped BUILD_ALL, which is normal and probably related how CMake generates  VS projects.

For the second issue, I would like you to ask to provide the wireshark trace file. Windows usually has issues with capturing its own traffic. Another issue which could be is, that OpENer needs raw sockets for UDP traffic, so if it is not started with administrator priviledges it could be that it fails.
I doubt that the sample application is not loaded, but I will take a look into that.

For the third issue I have to take a look first, but the Windows CI build works, so I can't give you a first idea how to fix that.

iceblu3710, 2020-05-14:
Mostly solved.
 - Running the program from VS as root just does not work. The firewall ports are still blocked. I need to attach to a running process instead.
 - Running the build from a separate command line as admin gets the proper UAC/Firewall prompts and communication working.

Issues:    
 - I can not enable tracing still. Makes logging difficult on windows.

CapXilinx, 2020-05-14:
I just gave it a try, Windows 10, VS 2017, and CMake 3.17.0 and traces works for me out of the box.

What do you mean with you define `OPENER_TRACE_LEVEL`? Thats turned on and off via a CMake switch

iceblu3710, 2020-05-14:
> What do you mean with you define OPENER_TRACE_LEVEL? Traces are turned on

What is the correct way to turn traces on? I added the define to the VS
pre-processor.

I have never used Cmake before, let alone on Windows.

CapXilinx, 2020-05-14:
The easiest way is probably to use the setup_windows.bat as explained in the README.md. Then set the "Where to build the binaries:" path in CMake to "bin/win32"
This should give you a predefined project. 
Then turn on the following switches:
[image]

Press configure and then generate

iceblu3710, 2020-05-15:
That solved my issues. I just learned about cmake-gui.

closed 2020-05-15 by iceblu3710