EIPStackGroup/OpENer #291: How to manually configure the IP

iceblu3710, 2020-05-12:
I am unsure how to properly load a custom IP like the documentation shows: `./opener ipaddress subnetmask gateway domainname hostaddress macaddress`

I am OK with doing this config in the init routine but HOW do I actually initialize the network structure with the proper info? The OpENer documentation is complete but very function literal without any context or tutorial beyond the sample application. I am having difficulty understanding how to adapt it.

CapXilinx, 2020-05-12:
Sorry, that a fragment from a former version. You do not configure the IP interface anymore that way.
Please take a look into the README.md how to do it with the current version

iceblu3710, 2020-05-14:
Any way to specify which multi-homed IP OpENer is to use when an interface has multiple one's set-up?

I see that OpENer mostly binds to the MAC and uses the default interface settings but I need the alternate addresses available.

I will need to make a sample project to see how windows reports multi-homing. So far it's pretty unclear and only a few programs know how to handle it.

azoitl, 2020-05-14:
I think what you like to have is way beyond the current target devices of OpENer. Also i see several conflicts in respect to the EtherNet/IP and CIP specs.

iceblu3710, 2020-05-14:
I don't know the specs so I'm not sure if I am asking too much. My goal is
to run an ENIP stack locally that ties into an in house SDK. We can use
that to train personnel with a "real world" interface.

The issue with that is all our laptops only have one NIC and we need to run
the SoftPLC with the ENIP scanner on that same NIC.

azoitl, 2020-05-14:
This screams for a VM or Docker solution.

iceblu3710, 2020-05-14:
Except that the SDK requires high level calls so a VM won't work.

azoitl, 2020-05-14:
SoftPLC in VM? So that you have a better predefined test environment.

closed 2021-11-02 by CapXilinx