EIPStackGroup/OpENer #310: Output-only connection point

nefethael, 2020-06-15:
Hi everyone,

I'm wondering if it's possible to have an "output-only" application type that would only retrieve I/O messages from scanner and send an heartbeat in return?

Does OpENer already support it?

Best regards,
Someone who's diving into CIP abyss

CapXilinx, 2020-06-15:
Hi,

to my understanding, this would still to be an exclusive owner connection, as output assemblys need an exclusive (or redundant) owner. If you do not need input information or there is no input information, then a dummy assembly can be used for that with a high RPI.

So yes, OpENer should be able to support this.

Best regards,
Martin

PS: Have fun, it is a hard, but rewarding ride ;)

nefethael, 2020-06-16:
@CapXilinx Thank's for the answer and encouragements :)

closed 2020-06-16 by nefethael

nefethael, 2020-07-01:
@CapXilinx Hi back, 

When setting up connection and with OPENER_PRODUCED_DATA_HAS_RUN_IDLE_HEADER option enabled, OpENer removes run/idle header size only if assembly is not an heartbeat here [cipioconnection.c](https://github.com/EIPStackGroup/OpENer/blob/master/source/src/cip/cipioconnection.c#L251), am I correct?

Shouldn't OpENer adds run/idle header only if assembly is not an heartbeat here: 
https://github.com/EIPStackGroup/OpENer/blob/master/source/src/cip/cipioconnection.c#L840

Best regards,
Vincent

CapXilinx, 2020-07-01:
Hi @nefethael 

I am not sure if I understand you correct. Modeless and Heartbeat shall not include run/idle information. When deactivating the compiler switch you will get a Modeless connection, otherwise it will be a 32-bit header connection. So this should be correct.

If the connection is a heartbeat, then the run/idle shall not be added.

Oh, do you mean because the information is overwritten if is shall not be present?

Best regards,
Martin