EIPStackGroup/OpENer #311: Status of EthernetLink Attributes 4,5,12,13 (Interface and media counters)

desmith2, 2020-06-30:
Hi,
We have several products using OpENer and are planning for future releases.

Question: I see that the Ethernet Link Object Attributes 4,5,12,13 (Interface and media counters) have been added to quite a bit from our current code. Are these completed or near completion?

Thanks, Dennis Smith

CapXilinx, 2020-07-10:
Sorry for the delay. I have to check, but I think it should be at least partly available. The code was provided by @stm-at-esd

CapXilinx, 2020-07-10:
@desmith2 on a different topic. You say you are using OpENer in products. May I ask which products and which company? It is always nice to show some success stories

Best regards,
Martin

desmith2, 2020-07-13:
Hi Martin, 

The company is Omron ODT (Delta Tau) division. The product is the Power PMAC motion controller.

http://www.deltatau.com/DT_PowerPMAC/PowerPMACHome.aspx

FYI: I have pushed for myself and others at Omron to contribute what ever we can to this project, and have been given the OK to offer such. It is a new addition to the product mentioned and market response is driving requests for specification completions.

I'll be applying thru the developer request area shortly.

The Best,
Dennis Smith
dennis.smith@omron.com