EIPStackGroup/OpENer #314: Overflow in HandleReceivedSendRequestResponseDataCommand (encape.c)

AlexNov1, 2020-07-20:
Hi,
I use the last version of OpENer V2.3. The problem i found is the following:

The return data type of function HandleReceivedSendRequestResponseDataCommand in encape.c is EipStatus, but it is not the status that is returned, but the size of the reply message.

Here the function with comments:

EipStatus HandleReceivedSendRequestResponseDataCommand(
  const EncapsulationData *const receive_data,
  const struct sockaddr *const originator_address,
  ENIPMessage *const outgoing_message) {
  EipStatus return_value = kEipStatusOkSend;                                        // <= DEFINITION OF RETURN VARIABLE

  if (receive_data->data_length >= 6) {
    /* Command specific data UDINT .. Interface Handle, UINT .. Timeout, CPF packets */
    /* don't use the data yet */
    GetDintFromMessage(
      (const EipUint8 **const ) &receive_data->current_communication_buffer_position);                            /* skip over null interface handle*/
    GetIntFromMessage(
      (const EipUint8 **const ) &receive_data->current_communication_buffer_position);                            /* skip over unused timeout value*/
    ( (EncapsulationData *const)receive_data )->data_length -= 6;             /* the rest is in CPF format*/

    if (kSessionStatusValid == CheckRegisteredSessions(receive_data) )            /* see if the EIP session is registered*/
    {
      EipInt16 send_size =
        NotifyCommonPacketFormat(receive_data,
                                 originator_address,
                                 outgoing_message);
      return_value = send_size;                                                                           // <= ASSIGN THE VALUE OF SEND SIZE TO RETURN VARIABLE THAT LEADS TO OVERFLOW

      if (send_size < 0) {                   /* need to send reply */
        return_value = kEipStatusError;
      }
    } else {             /* received a package with non registered session handle */
      InitializeENIPMessage(outgoing_message);
      GenerateEncapsulationHeader(receive_data,
                                  0,
                                  receive_data->session_handle,
                                  kEncapsulationProtocolInvalidSessionHandle,
                                  outgoing_message);
    }
  }
  return return_value;
}

CapXilinx, 2020-07-20:
Thanks!
I am currently fixing this kind of problems in the master branch. I will take a look if this is still in the code

AlexNov1, 2020-07-20:
All right, thank you!
I then close the issue.

Best regards
Alex

closed 2020-07-20 by AlexNov1