EIPStackGroup/OpENer #315: about multcast！

frankgp, 2020-08-01:
I try to port the OpENer into STM32 with LWIP. but I meet some problems.  
until now, the p2p mode is no problem, but for multicast, I meet some problems. 
I meet the LWIP not support problem.

It is about the generic_networkhandler.c
For CreateUdpSocket() function. 
about producing,   new_socket = socket(AF_INET, SOCK_RAW, IPPROTO_UDP);
and 
setsockopt(new_socket, IPPROTO_IP, IP_MULTICAST_IF,
                        NWBUF_CAST & my_addr.s_addr,
                        sizeof my_addr.s_addr )
for setsockopt(), the LWIP don't support it.

I don't know the "NWBUF_CAST & my_addr.s_addr" in detail.  
Do you have any suggestion for this problems?

frankgp, 2020-08-03:
When I ignore this error of this function.
setsockopt(new_socket, IPPROTO_IP, IP_MULTICAST_IF,
NWBUF_CAST & my_addr.s_addr,
sizeof my_addr.s_addr )

Now. STM32 + lwip can multicast T->0 with NJ1120 units.

closed 2020-08-03 by frankgp

CapXilinx, 2020-08-18:
If you do not join the multicast group, it could happen that some switches will not forward the packages to your port. It seems that there is a `igmp_joingroup ` function which handles this in LWIP