EIPStackGroup/OpENer #318: Linker Errors on Manjaro Linux

ergouser, 2020-08-30:
Manjaro Linux 
Linux 4.19.139-1-MANJARO #1 SMP Tue Aug 11 16:10:08 UTC 2020 x86_64 GNU/Linux

cmake version 3.18.1
gcc (GCC) 10.1.0
GNU ld (GNU Binutils) 2.34.0

_$git clone https://github.com/EIPStackGroup/OpENer.git
Cloning into 'OpENer'...
remote: Enumerating objects: 9, done.
remote: Counting objects: 100% (9/9), done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 147957 (delta 0), reused 1 (delta 0), pack-reused 147948
Receiving objects: 100% (147957/147957), 146.28 MiB | 3.77 MiB/s, done.
Resolving deltas: 100% (124628/124628), done.
$ cd OpENer/bin/posix/
$ ./setup_posix.sh 
-- The C compiler identification is GNU 10.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for srand
-- Looking for srand - found
-- Looking for rand
-- Looking for rand - found
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
Doxygen needs to be installed to generate the Doxygen documentation
-- No additional activated objects
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/opener/OpENer/bin/posix_

make builds source (with warnings) and returns these errors (complete build file is attached).
[opener1.txt](https://github.com/EIPStackGroup/OpENer/files/5147423/opener1.txt)

[ 97%] Linking C executable OpENer
/usr/bin/ld: ../../cip/libCIP.a(cipconnectionmanager.c.o):(.bss+0x0): multiple definition of `g_timestamps'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x0): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipconnectionmanager.c.o):(.bss+0x140): multiple definition of `master_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x140): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipconnectionmanager.c.o):(.bss+0x1c0): multiple definition of `read_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x1c0): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipconnectionmanager.c.o):(.bss+0x240): multiple definition of `highest_socket_handle'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x240): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipconnectionmanager.c.o):(.bss+0x244): multiple definition of `g_current_active_tcp_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x244): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipconnectionmanager.c.o):(.bss+0x250): multiple definition of `g_time_value'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x250): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipconnectionmanager.c.o):(.bss+0x260): multiple definition of `g_actual_time'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x260): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipconnectionmanager.c.o):(.bss+0x268): multiple definition of `g_last_time'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x268): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipconnectionmanager.c.o):(.bss+0x280): multiple definition of `g_network_status'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x280): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipioconnection.c.o):(.bss+0x0): multiple definition of `g_timestamps'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x0): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipioconnection.c.o):(.bss+0x140): multiple definition of `master_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x140): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipioconnection.c.o):(.bss+0x1c0): multiple definition of `read_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x1c0): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipioconnection.c.o):(.bss+0x240): multiple definition of `highest_socket_handle'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x240): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipioconnection.c.o):(.bss+0x244): multiple definition of `g_current_active_tcp_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x244): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipioconnection.c.o):(.bss+0x250): multiple definition of `g_time_value'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x250): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipioconnection.c.o):(.bss+0x260): multiple definition of `g_actual_time'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x260): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipioconnection.c.o):(.bss+0x268): multiple definition of `g_last_time'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x268): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipioconnection.c.o):(.bss+0x280): multiple definition of `g_network_status'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x280): first defined here
/usr/bin/ld: ../../enet_encap/libENET_ENCAP.a(encap.c.o):(.bss+0x0): multiple definition of `g_timestamps'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x0): first defined here
/usr/bin/ld: ../../enet_encap/libENET_ENCAP.a(encap.c.o):(.bss+0x140): multiple definition of `master_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x140): first defined here
/usr/bin/ld: ../../enet_encap/libENET_ENCAP.a(encap.c.o):(.bss+0x1c0): multiple definition of `read_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x1c0): first defined here
/usr/bin/ld: ../../enet_encap/libENET_ENCAP.a(encap.c.o):(.bss+0x240): multiple definition of `highest_socket_handle'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x240): first defined here
/usr/bin/ld: ../../enet_encap/libENET_ENCAP.a(encap.c.o):(.bss+0x244): multiple definition of `g_current_active_tcp_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x244): first defined here
/usr/bin/ld: ../../enet_encap/libENET_ENCAP.a(encap.c.o):(.bss+0x250): multiple definition of `g_time_value'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x250): first defined here
/usr/bin/ld: ../../enet_encap/libENET_ENCAP.a(encap.c.o):(.bss+0x260): multiple definition of `g_actual_time'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x260): first defined here
/usr/bin/ld: ../../enet_encap/libENET_ENCAP.a(encap.c.o):(.bss+0x268): multiple definition of `g_last_time'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x268): first defined here
/usr/bin/ld: ../../enet_encap/libENET_ENCAP.a(encap.c.o):(.bss+0x280): multiple definition of `g_network_status'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x280): first defined here
/usr/bin/ld: ../libPLATFORM_GENERIC.a(generic_networkhandler.c.o):(.bss+0x0): multiple definition of `g_timestamps'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x0): first defined here
/usr/bin/ld: ../libPLATFORM_GENERIC.a(generic_networkhandler.c.o):(.bss+0x140): multiple definition of `master_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x140): first defined here
/usr/bin/ld: ../libPLATFORM_GENERIC.a(generic_networkhandler.c.o):(.bss+0x1c0): multiple definition of `read_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x1c0): first defined here
/usr/bin/ld: ../libPLATFORM_GENERIC.a(generic_networkhandler.c.o):(.bss+0x240): multiple definition of `highest_socket_handle'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x240): first defined here
/usr/bin/ld: ../libPLATFORM_GENERIC.a(generic_networkhandler.c.o):(.bss+0x244): multiple definition of `g_current_active_tcp_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x244): first defined here
/usr/bin/ld: ../libPLATFORM_GENERIC.a(generic_networkhandler.c.o):(.bss+0x250): multiple definition of `g_time_value'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x250): first defined here
/usr/bin/ld: ../libPLATFORM_GENERIC.a(generic_networkhandler.c.o):(.bss+0x260): multiple definition of `g_actual_time'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x260): first defined here
/usr/bin/ld: ../libPLATFORM_GENERIC.a(generic_networkhandler.c.o):(.bss+0x268): multiple definition of `g_last_time'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x268): first defined here
/usr/bin/ld: ../libPLATFORM_GENERIC.a(generic_networkhandler.c.o):(.bss+0x280): multiple definition of `g_network_status'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x280): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [src/ports/POSIX/CMakeFiles/OpENer.dir/build.make:112: src/ports/POSIX/OpENer] Error 1
make[1]: *** [CMakeFiles/Makefile2:365: src/ports/POSIX/CMakeFiles/OpENer.dir/all] Error 2
make: *** [Makefile:103: all] Error 2

Any suggestions or ideas?

Thanks,

CapXilinx, 2020-09-02:
Checked this right now with Debian/GNU GCC 8.3.0 and didn't had any issue.

My best guess would be, that your toolchain has a problem with the extern keyword as only global variables are concerned.

CapXilinx, 2020-09-02:
"GCC now defaults to -fno-common. As a result, global variable accesses are more efficient on various targets. In C, global variables with multiple tentative definitions now result in linker errors. With -fcommon such definitions are silently merged during linking."

Perhaps this is the issue.  Could you check if external definitions are missing?

CapXilinx, 2020-10-15:
As there is no further reaction to this issue, I close it now

closed 2020-10-15 by CapXilinx

Broekman, 2020-12-06:
I had the same issue on Mingw, adding -fcommon resolved it for me.

hills86073, 2021-03-11:
Hi @Broekman I am running into the same multiple definition issue.  I am running Fedora 33 on a NUC (gcc 10.2.1) and would like to try adding -fcommon but I am not sure where to add it.  Any help much appreciated.

hills86073, 2021-03-12:
@Broekman , @CapXilinx 
Located a means to pass flag via CMakeCCompiler.cmake but when I try it does not resolve the issue.  
set(CMAKE_C_COMPILER "/usr/bin/gcc")
set(CMAKE_C_COMPILER_ARG1 "**_-fcommon_**")
I played around with this a bit by passing --verbose and -c to check functionality.  Both affected the make so that was promising but I am still stuck.

hills86073, 2021-03-13:
Got it working by adding __attribute__((__common__))  to the end of g_timestamps in generic_networkhandler.h
 
SocketTimer g_timestamps[OPENER_NUMBER_OF_SUPPORTED_SESSIONS]  __attribute__((__common__));