EIPStackGroup/OpENer #320: Implicit Messaging

AlexNov1, 2020-09-09:
Hi everyone,

I try to exchange the implicit messages and use the EEIP.NET library and the EIPScan tool for this. Unfortunately I haven't succeeded so far and so I hope for your tips.

Here are the configurations I've tried so far:

EEIP:NET:
[image]
Error: 'Connection failure, General Status Code: 1 Additional Status Code: 306 unknown'

EIPScan:
[image]

Does anyone have any idea what the problem is?

CapXilinx, 2020-09-09:
Hi.

> Error: 'Connection failure, General Status Code: 1 Additional Status Code: 306 unknown

Your device is sending a Null Forward Open, which is not supported by OpENer currently. It seems not to respect the EDS file in this regard.

Best regards,
Martin

CapXilinx, 2020-09-09:
Your EIPScan screenshot indicates, that the packages aren't reaching the OpENer device and are therefore unanswered

AlexNov1, 2020-09-10:
Hi Martin,
thanks for your quick reply.

I don't use the EDS file and configure the communication manually, as can be seen on the screenshots.
I have just looked at the trace log, whether with EEIP.NET library or EIPScan the OpENer reports the same:

networkhandler: new TCP connection
>>> network handler: accepting new TCP socket: 3 
networkhandler: opened new TCP connection on fd 3
Entering HandleDataOnTcpSocket for socket: 3
Data received on tcp:
Handles data for TCP socket: 3
Register session
Adds socket 3 to socket timers
Sets time stamp for socket 3
Sets time stamp for socket 3
TCP reply sent:
Sets time stamp for socket 3
TCP response was not fully sent
Entering HandleDataOnTcpSocket for socket: 3
Data received on tcp:
Handles data for TCP socket: 3
Send Request/Reply Data
Sockaddr type id: 32769
NotifyMessageRouter: routing unconnected message
NotifyMessageRouter: calling notify function of class 'connection manager'
notify: found instance 1
notify: calling ForwardOpen service
ForwardOpen: ConConnID 2750872067, ProdConnID 2750875646, ConnSerNo 2046
We have a Null request
We have a Non-Matching request
Right now we cannot handle Null requests
AssembleForwardOpenResponse: sending error response
Sets time stamp for socket 3
TCP reply sent:
Sets time stamp for socket 3

Do you have any tip for configuring the communication so that no null forward open is sent?
I'm just getting started with implicit messaging and I really don't know how to do it :(

CapXilinx, 2020-09-10:
That's up to the tool used. In principle, you do not need a null forward open for an implicit communication as all config data can be sent in a Non-Null Forward Open. The ENIPScanner tool for example sends no Null Forward Opens.

Null Forward Opens are used for pinging an ENIP device or to reconfigure a connection.

Best regards,
Martin

AlexNov1, 2020-09-10:
My problem is that I don't know how to switch from null forward open to non null forward open. Or is it permanently set in the tools so that I have no control over it?
I did not find the ENIPScanner tool. Is it possible for you to add the link to that tool here?

CapXilinx, 2020-09-10:
sorry, it is called ENIP Explorer https://sourceforge.net/projects/enipexplorer/

CapXilinx, 2020-09-10:
I do not know the tools you are using, so I cannot really help you there.

AlexNov1, 2020-09-11:
I have now loaded the Ethernet/IP Explorer (V1.2.0) and tried the implicit messaging with it. Unfortunately the result is the same as with my tools:
[image]

Wireshark also shows the same feedback from the OpENer for Ethernet/IP Explorer as with my tools:
[image]

Do you know where the mistake could be?

I'm sorry that I'm taking so much time away from you, but I just don't know how else to get to the solution.

CapXilinx, 2020-09-17:
Did you change anything on the OpENer code or config?
Sorry, I didn't had the time to try this myself

AlexNov1, 2020-09-18:
I only fixed the bug I told you about in issue #314.
As of now I would have to debug to find out what is going wrong. If you have a tip where I can specifically check the code, it would be really great :)

CapXilinx, 2020-09-18:
Just tested it, and everything works for me
[image]

CapXilinx, 2020-10-15:
As there is no further reaction to this issue, I close it now

closed 2020-10-15 by CapXilinx