EIPStackGroup/OpENer #321: Multiple exclusive owner connections in EDS file

svenhaitjema, 2020-09-11:
The description of OpENer states that multiple I/O connections are possible. I think I have setup the stack such that it should have two separate Exclusive Owner connections. 

Would it be possible to provide an additional EDS file that show's how to setup additional exclusive owner connections? At the moment when editing the EDS file by copying Connection1 and updating that to point to different assemblies, Studio 5000 only allows you to select one of the two when importing the Module, so I am assuming additional information / settings are needed. I could provide the sample_application.c that enables this functionality.

Thank you providing this Stack and for your time!

svenhaitjema, 2020-09-21:
Please let me know if I have confused terminology here that makes answering this question difficult/ impossible to answer or if this is not the right place to ask this question. 

As said I am trying to achieve a higher throughput from PLC to edge device. One of the information sources on this topic, in this case the RTAutomations blog on Ethernet/ip and big data (https://www.rtautomation.com/rtas-blog/ethernet-ip-big-data-part-2/) states the following:

' 
OPTION 3 – Use Multiple Assemblies & Multiple I/O Connections

There is no requirement in EtherNet/IP to have a single Input Assembly and a single Output Assembly. In fact, there are now provisions in ControlLogix and CompactLogix to support multiple assemblies and multiple I/O connections. 
'

I have setup the stack to have multiple assemblies and I/O connections, but I am failling to setup the EDS file correctly for this apparently. Any help or pointers in the right direction would be highly appreciated.

CapXilinx, 2020-09-22:
Hi @svenhaitjema

the source is correct, but a bit outdated (or ignoring something I will explain below) and providing more output assemblies is possible with OpENer. You have to modify the sample application to provide more and configure the EDS file accordingly. Unfortunately I am no expert on EDS files, so I cannot easily create a valid one for your case, but there is the EZ-EDS editor which should make life more easily.

OpENer supports the Large Forward Open Service, which allows larger I/O payloads than the usual 511 bytes, up to 65535 bytes.

svenhaitjema, 2020-09-24:
Dear @CapXilinx 

Thank you for your answer, the EZ-EDS does make life more easy :) I will check if it is possible to use the Large Forward Open Service on the PLC side. 

Your answer has allowed me to continue exploring, thank you so much!

closed 2020-09-24 by svenhaitjema

Suslsuk, 2023-03-14:
Hello, how did you manage to modifiy the EDS file so that it allows multible connections to one OpENer instance with an In/Out assembly for every connection? Our goal is to have one OpENer instance that serves as an adapter for up to 10 PLC's. Every PLC should have it's own Out/In assembly. So far we only managed to get the once connection with In/Out running for one Studio 5000 application.

Thanks so much

CapXilinx, 2023-03-21:
oh, and you need to change the limit of parallel exclusive owner connections

Suslsuk, 2023-03-27:
Thanks - I've done all that and I now have 2 exklusive connections with 3 assemblies each (in/out/config). It works well if only one of the two connections are used by a PLC, but as soon as a second connection is established most of the time both connections are closed with the following error message:
networkhandler: error on recv: 10035 - Ein nicht blockierender Socketvorgang konnte nicht sofort ausgeführt werden.