EIPStackGroup/OpENer #339: Conformance test CT18 failed using default configs

copercini, 2021-06-07:
Hi all, first thanks for the project, it's quite nice!

I'm porting the stack to my E/IP adapter project, but testing the example code with CT18 it's giving 200+ errors

Running the github untouched version directly on ubuntu (actually WLS2), and testing with latest OpENerPC.stc configs, it gives the same errors. My installation steps:
```
git clone https://github.com/EIPStackGroup/OpENer
cd OpENer/bin/posix
./setup_posix.sh
make (give some warnings but no errors)
sudo setcap cap_net_raw+ep ./src/ports/POSIX/OpENer
./src/ports/POSIX/OpENer eth0
```
At this point OpENer is running on ubuntu (exactly as on GitHub, no code changes), so downloading and editing [OpENerPC.stc](https://github.com/EIPStackGroup/OpENer/blob/master/data/OpENerPC.stc) with my local IP address and MAC, and running CT18 with this config in conformance mode, it's giving 242 errors and 9 warnings

I'm not sure if I can post the logs file here, but most errors are related to ForwardOpen and during Connection Manager test
```
ForwardOpen Returned unexpected Status = 0x1, Extended Status = 0x0000; expected Status = 0x0 (15 times this error)
Expected Successful ForwardOpen: Received General Status 01, Extended Status 0x0000 (almost 100 times this error)
Failed to open IO connection (5 times this error)
and some others
```

it looks similar to this issue: https://github.com/EIPStackGroup/OpENer/issues/188

am I doing something wrong in test? or can these errors be ignored?
I can try with different CT versions if needed

CapXilinx, 2021-06-08:
Hi,

I have CT-18 available, but only had time for a quick test run. I also get about that number of errors, some are related to, if I remember correct, GetAttributeSingle (and perhaps others) not or wrongly processing the electronic key segment. CT-17 runs without errors, so I guess additional tests have been added which need to be analyzed and addressed.

CapXilinx, 2021-06-09:
For the first error, check if the product code configured in CMake matches the one in the STC file.

For the second you have to check the box for the Type 1 Reset test

copercini, 2021-06-10:
Thanks, that worked! 
Checking the box for the "Type 1 Reset" test solved one issue and the another one was because I am using a gigabit network on test, and in STC file it's marked as 10/100mbps only, so also fixed, and it passes on CT-17 running on ubuntu!

 Now my next adventure is to fix the [Forward open problems](https://groups.google.com/g/eip-stack-group-opener-developers/c/420ZDjAty2A) in my LwIP port and pass the microcontroller on test

closed 2021-11-02 by CapXilinx