EIPStackGroup/OpENer #340: Add CMake find_package support OR documentation for non-source inclusion in projects

Justin-Davis, 2021-06-15:
Hello,

I am interested in using this repository for one of my projects, but would like to build it spearately and include the shared library build as a pre-built depenency. 

As of right now, to do so I need to manually specify the .so files and header folder locations to include. Since you are using CMake already, it would be very helpful if this repo shipped a package-config file which is compatible with CMake's [find_package](https://cmake.org/cmake/help/latest/command/find_package.html) command. 

Alternatively, some documentation on how to include the project after it has been built would be awesome. Right now, I am building with the option "-DCMAKE_INSTALL_PREFIX=/opener" and am copying the contents of that folder into my build environment. 

Thank you,
Justin Davis

Justin-Davis, 2021-06-15:
Note that packageable builds will also require some source changes -- the opener_api.h file looks for "ciptypes.h" which is in the cip folder. it cannot find it when included in this manner.

Justin-Davis, 2021-06-15:
`cmake --install .` also does not ship the 'devicedata.h' file, which would be necessary for compilation as well.

CapXilinx, 2021-06-16:
Thanks for the suggestions. Unfortunately, I know CMake well enough to provide a CMake package.
The "library mode" was added by @wpmccormick as far as I remember. I usually only use the pre-defined executable for development and tests.
Perhaps @wpmccormick can help you on how he uses the library in his projects.

CapXilinx, 2022-10-04:
Guess, this was addressed in  #375

If not feel free to reopen the issue

closed 2022-10-04 by CapXilinx