EIPStackGroup/OpENer #341: Random connection timeout

Mi5i, 2021-06-16:
Hi,

We have been experiencing some connection timeout with our application. In the end, we have tried to use only the unmodified sample application and the result was the same, connection timeouts. We have found no reason for this.
On the same network under the same master, we have another Ethernet/IP device and this device has no issue at all, everything works fine (the RPI for this device is 1ms).

For the sample application, we use 4ms RPI and we cannot set any lower value because the connection timeout occurs immediately, and we cannot set it any higher than 4ms, this is the top limit for our application.

We have captured network traffic with Wireshark, but we can only see graceful connection close and reopen from the OpENer side...
[image]
Scanner IP address: 192.168.213.148
OS: Windows 10 (version 1809)
OpENer version: 2.3.0

Please help,
Regards,
Mihaly

CapXilinx, 2021-06-16:
The standard OpENer time tick is 10ms, which is the lowest value the RPI could get in the default configuration.
If you want to go even lower and with reliable timings, you would have to at least use real-time kernel extensions and activate this configuration in CMake. Still not sure if the stack can provide that timings in its standard form

Mi5i, 2021-06-17:
Thank you very much for your help, it was very useful. We can close this issue.

closed 2021-06-17 by Mi5i