EIPStackGroup/OpENer #343: Implicit messaging : Udp listening Socket binds all network adapters

DevRga, 2021-06-22:
Hi,

When using OpENer for a windows software, it seems that the implicit messaging listening socket (consuming direction) is bound to all network adapters (in point to point mode), instead of the unique adapter from where the connection is originated.
Source :  cip/cipioconnection.c : 376/383
Is there a specific reason to that ? For my test I use a gethostname on the g_current_active_tcp_socket to bind only this adapter. It seems working well, but I can't be sure of any undesirable side effect, anybody can tell ?

On my "exotic" setup, two softwares presenting an Ethernet/IP adapter should run on the same system.

CapXilinx, 2021-06-22:
No, that's not on purpose and in fact wrong behavior. Implicit connections should only be available on the corresponding network
I am doing my test on Linux primarily and I usually assume only one Ethernet connection

DevRga, 2021-06-22:
Thanks for the reply, here's my patch for now : 
[Try-to-limit-bind-to-correct-adapter-when-instanciat.patch.txt](https://github.com/EIPStackGroup/OpENer/files/6692899/Try-to-limit-bind-to-correct-adapter-when-instanciat.patch.txt)

CapXilinx, 2021-06-23:
Thanks, I will try it out asap

CapXilinx, 2021-07-20:
Added in the above-mentioned commit.
Thanks, @DevRga

commit 8fd79d3d (2021-07-20): Fixes #343 - binds UDP only to one interface

UDP connection socket, should only be created for the interface
which received the forward open request
closed 2021-07-20 by CapXilinx