EIPStackGroup/OpENer #344: linker errors when building.

rednexela1941, 2021-07-04:
Hi there,
Using the instructions in the README, I run:
`cd bin/posix`
`./setup_posix.sh`
`make`

And get the following issues (among many compiler warnings -- that I can include if that will be useful):

```
[100%] Linking C executable OpENer
/usr/bin/ld: ../../cip/libCIP.a(cipconnectionmanager.c.o):(.bss+0x0): multiple definition of `g_timestamps'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x0): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipconnectionmanager.c.o):(.bss+0x140): multiple definition of `master_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x140): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipconnectionmanager.c.o):(.bss+0x1c0): multiple definition of `read_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x1c0): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipconnectionmanager.c.o):(.bss+0x240): multiple definition of `highest_socket_handle'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x240): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipconnectionmanager.c.o):(.bss+0x244): multiple definition of `g_current_active_tcp_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x244): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipconnectionmanager.c.o):(.bss+0x250): multiple definition of `g_time_value'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x250): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipconnectionmanager.c.o):(.bss+0x260): multiple definition of `g_actual_time'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x260): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipconnectionmanager.c.o):(.bss+0x268): multiple definition of `g_last_time'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x268): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipconnectionmanager.c.o):(.bss+0x280): multiple definition of `g_network_status'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x280): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipioconnection.c.o):(.bss+0x0): multiple definition of `g_timestamps'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x0): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipioconnection.c.o):(.bss+0x140): multiple definition of `master_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x140): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipioconnection.c.o):(.bss+0x1c0): multiple definition of `read_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x1c0): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipioconnection.c.o):(.bss+0x240): multiple definition of `highest_socket_handle'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x240): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipioconnection.c.o):(.bss+0x244): multiple definition of `g_current_active_tcp_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x244): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipioconnection.c.o):(.bss+0x250): multiple definition of `g_time_value'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x250): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipioconnection.c.o):(.bss+0x260): multiple definition of `g_actual_time'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x260): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipioconnection.c.o):(.bss+0x268): multiple definition of `g_last_time'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x268): first defined here
/usr/bin/ld: ../../cip/libCIP.a(cipioconnection.c.o):(.bss+0x280): multiple definition of `g_network_status'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x280): first defined here
/usr/bin/ld: ../../enet_encap/libENET_ENCAP.a(encap.c.o):(.bss+0x0): multiple definition of `g_timestamps'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x0): first defined here
/usr/bin/ld: ../../enet_encap/libENET_ENCAP.a(encap.c.o):(.bss+0x140): multiple definition of `master_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x140): first defined here
/usr/bin/ld: ../../enet_encap/libENET_ENCAP.a(encap.c.o):(.bss+0x1c0): multiple definition of `read_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x1c0): first defined here
/usr/bin/ld: ../../enet_encap/libENET_ENCAP.a(encap.c.o):(.bss+0x240): multiple definition of `highest_socket_handle'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x240): first defined here
/usr/bin/ld: ../../enet_encap/libENET_ENCAP.a(encap.c.o):(.bss+0x244): multiple definition of `g_current_active_tcp_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x244): first defined here
/usr/bin/ld: ../../enet_encap/libENET_ENCAP.a(encap.c.o):(.bss+0x250): multiple definition of `g_time_value'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x250): first defined here
/usr/bin/ld: ../../enet_encap/libENET_ENCAP.a(encap.c.o):(.bss+0x260): multiple definition of `g_actual_time'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x260): first defined here
/usr/bin/ld: ../../enet_encap/libENET_ENCAP.a(encap.c.o):(.bss+0x268): multiple definition of `g_last_time'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x268): first defined here
/usr/bin/ld: ../../enet_encap/libENET_ENCAP.a(encap.c.o):(.bss+0x280): multiple definition of `g_network_status'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x280): first defined here
/usr/bin/ld: ../libPLATFORM_GENERIC.a(generic_networkhandler.c.o):(.bss+0x0): multiple definition of `g_timestamps'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x0): first defined here
/usr/bin/ld: ../libPLATFORM_GENERIC.a(generic_networkhandler.c.o):(.bss+0x140): multiple definition of `master_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x140): first defined here
/usr/bin/ld: ../libPLATFORM_GENERIC.a(generic_networkhandler.c.o):(.bss+0x1c0): multiple definition of `read_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x1c0): first defined here
/usr/bin/ld: ../libPLATFORM_GENERIC.a(generic_networkhandler.c.o):(.bss+0x240): multiple definition of `highest_socket_handle'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x240): first defined here
/usr/bin/ld: ../libPLATFORM_GENERIC.a(generic_networkhandler.c.o):(.bss+0x244): multiple definition of `g_current_active_tcp_socket'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x244): first defined here
/usr/bin/ld: ../libPLATFORM_GENERIC.a(generic_networkhandler.c.o):(.bss+0x250): multiple definition of `g_time_value'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x250): first defined here
/usr/bin/ld: ../libPLATFORM_GENERIC.a(generic_networkhandler.c.o):(.bss+0x260): multiple definition of `g_actual_time'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x260): first defined here
/usr/bin/ld: ../libPLATFORM_GENERIC.a(generic_networkhandler.c.o):(.bss+0x268): multiple definition of `g_last_time'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x268): first defined here
/usr/bin/ld: ../libPLATFORM_GENERIC.a(generic_networkhandler.c.o):(.bss+0x280): multiple definition of `g_network_status'; CMakeFiles/OpENer.dir/main.c.o:(.bss+0x280): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [src/ports/POSIX/CMakeFiles/OpENer.dir/build.make:106: src/ports/POSIX/OpENer] Error 1
make[1]: *** [CMakeFiles/Makefile2:350: src/ports/POSIX/CMakeFiles/OpENer.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

rednexela1941, 2021-07-05:
Update -- it seems one needs to select a target:
```
$ make help
The following are some of the valid targets for this Makefile:
... all (the default if no target is provided)
... clean
... depend
... edit_cache
... rebuild_cache
... doc
... CIP
... ENET_ENCAP
... NVDATA
... OpENer
... PLATFORM_GENERIC
... POSIXPLATFORM
... SAMPLE_APP
... Utils
```
(Although same error when building the OpENer target) -- do any examples of usage exist? -- SAMPLE_APP outputs an archive file -- and I'm not really sure why it is done this way.

rednexela1941, 2021-07-05:
Looks like this was related to a change introduced in gcc 10, which now defaults to -fno-common (https://gcc.gnu.org/gcc-10/changes.html). Making the following change (setting the -fcommon flag) made everything work:

```
diff --git a/source/buildsupport/POSIX/OpENer_PLATFORM_INCLUDES.cmake b/source/buildsupport/POSIX/OpENer_PLATFORM_INCLUDES.cmake
index 2943728e3..eaed567f5 100644
--- a/source/buildsupport/POSIX/OpENer_PLATFORM_INCLUDES.cmake
+++ b/source/buildsupport/POSIX/OpENer_PLATFORM_INCLUDES.cmake
@@ -1,6 +1,6 @@
 macro(opener_platform_spec)
   include_directories(${PORTS_SRC_DIR}/${OpENer_PLATFORM} ${PORTS_SRC_DIR}/${OpENer_PLATFORM}/sample_application)
-  set( CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -std=c99 -Werror=implicit-function-declaration" )
+  set( CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -std=c99 -Werror=implicit-function-declaration -fcommon" )
   if(NOT CMAKE_INSTALL_LIBDIR)
     set( CMAKE_INSTALL_LIBDIR "lib")
   endif()
```

CapXilinx, 2021-07-20:
Thanks.
Patch applied in commit 11550725013bcc5f60dc17ef47e044c66de71823

closed 2021-07-20 by CapXilinx