EIPStackGroup/OpENer #352: Ethernet IP simulation using OpENer

aaronrenny, 2021-12-14:
Hi all,

I am relatively new to ethernet IP and was looking for Linux solutions to simulate a ethernet IP network when I came across OpENer. I followed the steps given in the readme.md and was able to compile it. 
But after executing the command ./src/ports/POSIX/OpENer enp0s3; nothing was displayed on my screen. Is this the expected behaviour?
I would also like to know which code is running after i execute this command and where it is located so that i can modify it.

Also, I am looking for a ethernet IP scanner to work with the OpENer. Till now I have come across 
EIPScanner: https://github.com/nimbuscontrols/EIPScanner.git
can I use this to simulate a scanner on linux pc that works with the OpENer or are there any other options to do the same.
Please note that I donot have a PLC to work with and so I plan on simulating the scanner and the adapter in the same (or two different) PCs . Please do tell me if this is possible. Just the transfer of data through explicit messaging will suffice. I will attach the terminal output of the ./src/ports/POSIX/OpENer enp0s3 command
and also the dmesg output. Please do help me out.

Thank you,
Aaron 

[image]
[image]

jpbedrock, 2021-12-20:
Hi Aaron, I am no expert, but I will share what I have learned so far: the expected behavior is that there is nothing printed after launching OpENer.  It does not output anything on its own.  You will need a scanner to communicate with it.  The OpENer example copies data it receives and writes it back out, which you can try with the scanner.  I use the Hilscher scanner in windows.  Scan for devices on your network by clicking the Send List Identity Request button, then highlight the OpENer device.  Click on the Expl. Msg tab, and select Set_Attribute_single.  Use Class: 4 , Instance: 150, and Attribute: 3.  The OpENer sample program creates 2 32 byte structures, so fill the payload with 32 bytes of hex data, and click send.  The status at the top should turn green if successful.  Opener will automatically copy the data to the other structure.  Next, to read the data structure, do a Get_Attribute_Single, Class: 4, Instance: 100 Attr: 3.  Send.  The response fields should contain the data you sent with the set command.  

Hope this gets you running.

The source of the opener sample application is /OpENer/source/src/ports/POSIX/sample_application/sampleapplication.c  Which is the code that sets up the virtual device reflected in the .eds file located in the /OpENer/data directory.  The actual  OpENer source code is in the directories above sample_application.    You can modify this code and then use the same commands to make and launch OpENer and that will include your changes.

balaji-6975, 2022-03-22:
Hi Aaron,
I am new to Ethernet/IP to communicate plc using OpENer ,I am also faced same issue .is there any solution to transfer data to plc ,please guide me to overcome this issue

CapXilinx, 2022-03-22:
I don't know Nimbuscontrols ENIPScanner, but assuming it works as intended it should be able to create a connection with OpENer. There is also the ENIPTool from Hilscher which can be used to test OpENer or ENIPExplorer on Sourceforge, which I myself used and it worked with OpENer

closed 2022-03-22 by CapXilinx

balaji-6975, 2022-03-22:
Hi [CapXilinx](https://github.com/CapXilinx),
is there any c++ package to implement communication with OpENer

CapXilinx, 2022-03-23:
Hi @balaji-6975
last answer here, as this issue shouldn't be hijacked.
If you mean you need a C++ source for the functionality OpENer provides, then you can take a look on CIPster, which started as a C++ fork or OpENer. Not sure how the state of development is there, but to use the OpENer sources in a C++ project, it isn't necessary to convert it to C++ (check the Internet on that, there should be enough infos on how to use C sources in a C++ project).
If you need a scanner (client), then take a look at Nimbuscontrol's ENIPScanner.

balaji-6975, 2022-03-23:
Hi CapXilinx,
Thank you for suggesting C++ package ,I will try to make implementation on it.

nefethael, 2022-03-23:
> Hi all,
>
> I am relatively new to ethernet IP and was looking for Linux solutions to simulate a ethernet IP network when I came across OpENer. I followed the steps given in the readme.md and was able [...]

Yes it is possible, we do this at work

CapXilinx, 2022-03-23:
@aaronrenny  if you want some output you need to enable the traces in the CMake config and recompile. In normal operation, no output is necessary, as it is planned that OpENer is part of an embedded system (I/O module, power meter, ....)

balaji-6975, 2022-03-23:
Hi nefethael,
Can u give some more details about EDS file used to connect with EIPScanner,by default I am using OpENer EDS file in plc then Iam facing related to unknown distination epath.
Please provide some information to execute communication successful.
Thank you in advance.

CapXilinx, 2022-03-23:
@balaji-6975 please stop hijacking this issue!