EIPStackGroup/OpENer #353: Multiple OpENer Instances

jpbedrock, 2021-12-15:
I have been using OpENer for a few weeks and I have a single instance in Linux communicating with a scanner on another PC.  I now need to scale up and have a larger number (64-128) of OpENer instances running at the same time on the network in order to test simultaneous connections to the scanner.  Looking at the source code, OpENer binds to the mac address of the ethernet interface, which would mean I would need a separate ethernet interface for each OpENer instance.  After going down a few unsuccessfully paths such as aliased interfaces and multiple virtual interfaces in a VM (same subnet on multiple interfaces is bad) I would like to ask if there are any suggestions on how to get more than one instance working on a single PC or VM.  Thank you for any help

jpbedrock, 2022-03-16:
If anyone in the future is looking to do something similar, my workaround was to create a Docker image for OpENer and connect them to a Docker network in macvlan mode.  Macvlan will generate a unique mac for each OpENer instance allowing one machine to host as many OpENers as you want through a single enet (I have gone up to 128 so far) .

CapXilinx, 2022-03-23:
@jpbedrock the usual scenario for OpENer is one instance per device.
Your solution with Docker containers is a very nice approach. Was it complicated to create a docker config for it? I am not a docker expert, but I think it would be a nice addition for the project.

jpbedrock, 2022-06-14:
Yes, that is fine by me.  Thank you for asking.

commit 4cefc116 (2022-10-04): Closes #353 - Adds description to run multiple OpENers

with docker.
closed 2022-10-04 by CapXilinx