EIPStackGroup/OpENer #359: Electronic Key is not being verified in UCMM requests

copercini, 2022-02-03:
Hi,

I'm testing OpENer in CT18, and one of problems is that Electronic Key Format 4 is not being verified in Unconnected Message Manager requests, others requests seems fine

I'm trying to find where these UCMM requests go and if it's missing a `CheckElectronicKeyData()` somewhere, but I'm still learning about OpENer and EIP, any ideas/suggestions?

CapXilinx, 2022-03-07:
Hi, thanks for the report
sorry for the late answer, as I have been sick.
I would need to check the concrete error message, but probably it will miss the mentioned function somewhere and an appropriate response

lucinama, 2024-02-20:
@copercini Did you manage to overcome this issue? I'm encountering the same problem with CT20 and it does not seem to be fix, can you give me some tips ? Or @MartinMelikMerkumians ?
It would be very much appreciated.

copercini, 2024-02-21:
Hi/Olá @lucinama 
Unfortunately, I couldn't find a solution, so we ended up using a different EtherNet/IP solution than OpENer to pass the latest Conformance test

MartinMelikMerkumians, 2024-03-22:
Hi.

Well, the solution is to add check in UCMM request, as no further feedback was provided and I didnt had the time to investigate it wasn't solved yet