EIPStackGroup/OpENer #369: To use multiple connectionID's on ForwardOpen

rakesh1225, 2022-04-08:
Hello

We are using Opener version 2.1.0 on our Linux device. We are able to use it properly for sending/receiving single frame of message with single connection ID each.
We have a requirement to send 3 frames of messages with different connection ID's for each frame of messages.

Please let me know if we have to increase the number of ForwardOpens? 
What changes may be required in configuring the Opener stack.
Do I need to add more assembly messages from Openerstack like to increase the below code:

   // create 3 assembly object instances
    auto *assembly_data064 = cip_app.GetAssemblyData064();
    auto *assembly_data096 = cip_app.GetAssemblyData096();
    auto *assembly_data097 = cip_app.GetAssemblyData097();
    auto *assembly_data09A = cip_app.GetAssemblyData09A();

    // INPUT
    CreateAssemblyObject(DEMO_APP_INPUT_ASSEMBLY_NUM,
                         assembly_data064,
                         INPUT_ASSEMBLY_SIZE);

    // OUTPUT
    CreateAssemblyObject(DEMO_APP_OUTPUT_ASSEMBLY_NUM,
                         assembly_data096,
                         OUTPUT_ASSEMBLY_SIZE);

    // CONFIG
    CreateAssemblyObject( DEMO_APP_CONFIG_ASSEMBLY_NUM,
                         assembly_data097,
                         sizeof(assembly_data097));

CapXilinx, 2022-04-08:
Hi,
I don't understand your requirement. Could you please try to provide an example/figure of it.

Connections IDs are not chosen by the adapter, but by the scanner afaik

Best regards,
Martin

CapXilinx, 2022-04-08:
@rakesh1225 OpENer also supports the LargeForwardOpen, which allows larger payloads.

If LargeForwardOpen is not an option, you can add additional assemblies and create Input-only connections to them (by selecting the Input Only assembly as O->T target)

rakesh1225, 2022-04-08:
Yes, planning to use ForwardOpen only.

Does Input-only connections allow mulitple assemblies? As per the following function definition it allows only 0 as connection number and stores in array. It is discarding anything more than that.
ConfigureInputOnlyConnectionPoint(0,
                                      DEMO_APP_HEARTBEAT_INPUT_ONLY_ASSEMBLY_NUM,
                                      DEMO_APP_INPUT_ASSEMBLY_NUM1,
                                      DEMO_APP_CONFIG_ASSEMBLY_NUM);
ConfigureInputOnlyConnectionPoint(1,
                                      DEMO_APP_HEARTBEAT_INPUT_ONLY_ASSEMBLY_NUM,
                                      DEMO_APP_INPUT_ASSEMBLY_NUM2,
                                      DEMO_APP_CONFIG_ASSEMBLY_NUM);
ConfigureInputOnlyConnectionPoint(2,
                                      DEMO_APP_HEARTBEAT_INPUT_ONLY_ASSEMBLY_NUM,
                                      DEMO_APP_INPUT_ASSEMBLY_NUM3,
                                      DEMO_APP_CONFIG_ASSEMBLY_NUM);

[image]

CapXilinx, 2022-04-08:
No, you can only target one assembly with each connection. There is no way around that.
You would need 3 Input Assemblies and 1 output assemblies in your case, which means you will create one Exclusive Owner connections, and two additional Input-Only connection to get all of your data.

For that you have to tune the defines in the user conf

azoitl, 2022-04-08:
The CIP protocol does not really allow what you would like to have. If you have three assemblies to be sent from your device to the scanner you need to have three forward opens. However what you could have is that you have one normal input/output connection in your example 101 & 100 would be as @CapXilinx  just wrote be the exclusive owner connection and you could have two additional input only connections. However for the input only connection you also need recieving assemblies. These receiving assemblies can AFAIR be of lenght 0 but CIP need it to identify the connection. The receiving assemblies are used for the heart beat messages which have to be sent from the scanner to your devices. This is used by OpENer to detect timeouts like when the scanner is gone or cables are broken. I hope this helps.

rakesh1225, 2022-04-13:
Hello @CapXilinx , @azoitl 
Thanks for the response.

I will do 3 ForwardOpen's from Scanner.
So the following changes will be done from Adapter/Opener. However I don't find a way to send DEMO_APP_HEARTBEAT_INPUT_ONLY_ASSEMBLY_NUM. 

Will there be any issue to connection if HEARTBEAT will not be sent from Adapter to Scanner? or any other way to send HEARTBEAT from Adapter to scanner.

   ConfigureExclusiveOwnerConnectionPoint(0,
                                           DEMO_APP_OUTPUT_ASSEMBLY_NUM,
                                           DEMO_APP_INPUT_ASSEMBLY_NUM_1,
                                           DEMO_APP_CONFIG_ASSEMBLY_NUM);

    ConfigureInputOnlyConnectionPoint(0,
                                      DEMO_APP_HEARTBEAT_INPUT_ONLY_ASSEMBLY_NUM,
                                      DEMO_APP_INPUT_ASSEMBLY_NUM_2,
                                      DEMO_APP_CONFIG_ASSEMBLY_NUM);

    ConfigureInputOnlyConnectionPoint(1,
                                      DEMO_APP_HEARTBEAT_INPUT_ONLY_ASSEMBLY_NUM,
                                      DEMO_APP_INPUT_ASSEMBLY_NUM_3,
                                      DEMO_APP_CONFIG_ASSEMBLY_NUM);

CapXilinx, 2022-04-13:
The heartbeat sends no payload data. It only ensures that the other side is still alive

azoitl, 2022-04-13:
In your case the heartbeat is sent from the scanner to the adapter. In the other direction you have the input only data. However you rconfiguration will not work in the intended way. You need at least 2 different Heartbeat input only assemblies.

rakesh1225, 2022-04-28:
Thank you so much for the support. @azoitl, @CapXilinx 
Since the assembly data size is keep increasing as per our requirement, we are trying to use large forward open instead of multiple forward open's.

Large forward open is working good with assembly data  size less than 1444 only. But when I tried to increase the large forward open size to 1603 bytes, it is failing to send data.

Note: I increased the macro PC_OPENER_ETHERNET_BUFFER_SIZE size to 1603 already. Still couldn't send data size of 1603 over CIP using large forward open.

Pl confirm if any more changes required to increase large forward open size in opener stack.

CapXilinx, 2022-06-14:
As there has been no respose anymore, I assue this is fixed

closed 2022-06-14 by CapXilinx