EIPStackGroup/OpENer #379: Potential bugs

widavies, 2022-06-23:
Shouldn't [this line](https://github.com/EIPStackGroup/OpENer/blob/cef2833fcf2ea9fda68f85c4638429956128daf5/source/src/cip/cipqos.c#L75)
be:
```c
const EipUint8 ** cip_message = &message_router_request->data;
```
instead of:
```c
const EipUint8 ** cip_message = message_router_request->data;
```

Additionally, the same case for [this line](https://github.com/EIPStackGroup/OpENer/blob/cef2833fcf2ea9fda68f85c4638429956128daf5/source/src/cip/cipcommon.c#L967)

Thanks

CapXilinx, 2022-06-24:
Hi,

yes, I think you are right. Seems it is "fixed" by later using the & operator on it...?

Thanks, would be great if you can provide a PR.

Best regards

widavies, 2022-06-24:
Sure, I'd love to. 

One thing to clarify first - You create a double pointer to the uint8_t data in `message_router_request->data`.

My understanding is that as you decode data, you extract the data data from the buffer, then increment the buffer forward. 

For example:
```
const EipUint8 ** cip_message = &message_router_request->data;
*cip_message += string->length
```

I've been staring at this line for awhile, but wouldn't it technically be equivalent to just:
```
message_router_request += string->length
```

I'm not sure if instead the double pointer is supposed to get incremented forward or not, but some insight on this would be great.

Thanks for the great library, just ported to LWIP/no-RTOS on ARM and it's working smoothly.

CapXilinx, 2022-07-22:
Hmm, I think you are absolutely right.
Could be that when I changed stuff on the MRR implementation, I went with the fastest, but probably not the clearest way to update the code 😅

widavies, 2022-08-01:
No worries, it doesn't look like the functions are even used anyway, they're more just helper functions than anything.

jvalenzuela, 2022-10-08:
This has been corrected in the set of commits I'm submitting for addressing compiler warnings, and is also a rather good example why warnings should not be ignored and allowed to accumulate. Here the compiler identified an actual problem, but was masked by the overwhelming number of nuisance warnings.

CapXilinx, 2023-05-06:
Should be solved by now

closed 2023-05-06 by CapXilinx