EIPStackGroup/OpENer #390: malformed sendRR data packet on multicast forward open reply

cperrysfl, 2022-08-26:
After some time (not 100% sure about the pre-conditions), my OpENer stack produces a malformed sendRR encapsulation layer when generating a forward open response with a multicast T->O endpoint.

[image]

I've Identified some problematic circumstances in `OpenProducingMulticastConnection` in `cipioconnection.c`. The code expects one of the two elements of the `g_common_packet_format_data_item.address_info_item` array to have a 0 type_id and use it to store the T->O socket informations:

```c
int j = 0; /* allocate an unused sockaddr struct to use */
if(g_common_packet_format_data_item.address_info_item[0].type_id == 0) { /* it is not used yet */
  j = 0;
} else if(g_common_packet_format_data_item.address_info_item[1].type_id ==
          0) {
  j = 1;
}
```

When the bug happens, both index 0 and 1 have a non-zero `type_id`. When this happens, the `OpenProducingMulticastConnection` function just keeps on with `j=0`, which is the default value and end up overwriting the O->T socket info.

Later on, when generating the response in `AssembleLinearMessage` in `cpf.c`, the function cannot find the O->T socket information in the final double for loop, which results in a malformed packet (item_count was set to 4 but only 3 items actually get added to the message).

capture shown in the screenshot: [opener-malformed-sendrr-last-packet.zip](https://github.com/EIPStackGroup/OpENer/files/9435908/opener-malformed-sendrr-last-packet.zip)

cperrysfl, 2022-09-22:
I think I've identified the root cause of this problem, it comes down to how many Command Specific Data items are sent to the adapter when the scanner makes the forward open request.

Most of the scanner tools I tested (EIPScanner and EnIPExplorer) only send 2 command specific data, a "Null Address Item" (0x0000) and an "Unconnected Data Item" (0x00b2).

Some scanner will send 3 command specific data, both the "Null Address Item", the "Unconnected Data Item" and the "T->O socket address info" (0x8001) so that the adapter knows which multicast address to use rather than the default values in the TCP/IP object.

If the scanner sends 4 command specific data (all 3 stated above and the "O->T socket address info" (0x8000)), the bug described in this issue will happen.

So to reproduce this bug, one needs to make a forward open request with a multicast T->O endpoint and make sure the request contains both a "T->O socket address info" (0x8001) and a "O->T socket address info" (0x8000). This is something ODVA CT18.1 will do (if the adapter supports multicast).

CapXilinx, 2022-10-03:
Thanks for your patience and your fix. I am currently reviewing the PR