EIPStackGroup/OpENer #396: I transplanted the protocol stack to STM32, and there were many obvious errors in the source code

hangeV, 2022-09-15:
I transplanted the protocol stack to STM32, and there were many obvious errors in the source code。
[image]
what is the reason?

CapXilinx, 2022-09-15:
The reason for what?

hangeV, 2022-09-15:
> The reason for what?

int DecodeCipAssemblyAttribute3(CipByteArray *const data,
                                const CipMessageRouterRequest *const message_router_request,
                                CipMessageRouterResponse *const message_router_response)
{

  const EipUint8 **const cip_message = message_router_request->data;

  CipInstance *const instance =
    GetCipInstance(GetCipClass(
                     message_router_request->request_path.class_id),
                   message_router_request->request_path.instance_number);

  int number_of_decoded_bytes = -1;
  OPENER_TRACE_INFO(" -> set Assembly attribute byte array\r\n");
  CipByteArray *cip_byte_array = data;

  if(message_router_request->request_data_size < data->length) {
    OPENER_TRACE_INFO(
      "DecodeCipByteArray: not enough data received.\n");
    message_router_response->general_status = kCipErrorNotEnoughData;
    return number_of_decoded_bytes;
  }
  if(message_router_request->request_data_size > data->length) {
    OPENER_TRACE_INFO(
      "DecodeCipByteArray: too much data received.\n");
    message_router_response->general_status = kCipErrorTooMuchData;
    return number_of_decoded_bytes;
  }

  // data-length is correct
  memcpy(cip_byte_array->data, cip_message, cip_byte_array->length);

  if(AfterAssemblyDataReceived(instance) != kEipStatusOk) {
    /* punt early without updating the status... though I don't know
     * how much this helps us here, as the attribute's data has already
     * been overwritten.
     *
     * however this is the task of the application side which will
     * take the data. In addition we have to inform the sender that the
     * data was not ok.
     */
    message_router_response->general_status = kCipErrorInvalidAttributeValue;
  } else {
    message_router_response->general_status = kCipErrorSuccess;
  }

  number_of_decoded_bytes = cip_byte_array->length;

  return number_of_decoded_bytes;
}

for example:
              const EipUint8 **const cip_message = message_router_request->data;
should   const EipUint8 *const cip_message = message_router_request->data;

CapXilinx, 2022-09-15:
Because sometimes, the data pointer is a pointer to the actual data structure.

If you think these are programming errors, I am happy to receive a patch. Currently the code is working for me, and, as you posted screenshots from your diff tool, it is hard for me to even find the code sections you deem to be erroneous.

hangeV, 2022-09-16:
> Because sometimes, the data pointer is a pointer to the actual data structure.
>
> If you think these are programming errors, I am happy to receive a patch. Currently the code is working for me, and, [...]

"cipassembly.c" File, "DecodeCipAssemblyAttribute3" Function, 
const EipUint8 **const cip_message = message_router_request->data;
typedef struct {
  CipUsint service;
  CipEpath request_path;
  EipInt16 request_data_size;
  const CipOctet *data;
} CipMessageRouterRequest;
typedef uint8_t CipOctet;

hangeV, 2022-09-19:
int DecodeCipString(CipString *const data,
                    const CipMessageRouterRequest *const message_router_request,
                    CipMessageRouterResponse *const message_router_response) {
  printf("DecodeCipString()\n");
  const EipUint8 *const cip_message = message_router_request->data;

  int number_of_decoded_bytes = -1;
  CipString *string = data;
  string->length = GetIntFromMessage((const EipUint8 **const)&cip_message);
  memcpy(string->string, cip_message, string->length);
  *cip_message += string->length;

  number_of_decoded_bytes = string->length + 2; /* we have a two byte length field */
  if(number_of_decoded_bytes & 0x01) {
    /* we have an odd byte count */
    ++(*cip_message);
    number_of_decoded_bytes++;
  }
  message_router_response->general_status = kCipErrorSuccess;
  return number_of_decoded_bytes;
}

this function also error，const EipUint8 *const cip_message = message_router_request->data;，*cip_message += string->length;

closed 2022-12-13 by CapXilinx