EIPStackGroup/OpENer #406: Are there any examples?

themightyoarfish, 2022-10-21:
Does there exist any sample code for simple use cases such as polling an Ethernet/IP device?
I couldn't find anything instructive in the autogenerated doc.

Suslsuk, 2022-12-22:
Is there a way to get a Visual Studio project to get started. For me as a pure Windows developer it is difficult to start with OpENer.

For example, it would be great to see sample code that other developers have used to implement an OpENer Implicit IO messaging adapter. 

Thanks

azoitl, 2022-12-22:
The Cmake build system we are using allows to generate a VS project. I think this could help to get you started. The default example is already an Implicit IO messaging adapter.

Suslsuk, 2022-12-22:
You are right, Cmake created the VS project. Thanks!

nvienhuy, 2023-01-28:
Hi! How exactly do I create a VS project? Sorry but I am new and just trying to learn a thing or two, but could any one please instruct me on how to do that?

nvienhuy, 2023-01-28:
Hi! How exactly do I create a VS project for the sample application? Sorry I am new and just trying to learn a thing or two, but could any one please instruct me on how to do that?

azoitl, 2023-01-28:
Hi as pointed out above we are using CMake as make file and project generation tool. When you use CMake gui one of the first questions CMake will ask you what kind of project to generate. Here you can select visual studio and with that you should be set. For more info please refer to the CMake documentation.

CapXilinx, 2023-08-16:
Closed, seems solved to me

closed 2023-08-16 by CapXilinx