EIPStackGroup/OpENer #447: Two or more implicit connections at the same time

Suslsuk, 2023-04-20:
### Discussed in https://github.com/EIPStackGroup/OpENer/discussions/440

<div type='discussions-op-text'>

<sup>Originally posted by **Suslsuk** April  4, 2023</sup>
[sample_application.zip](https://github.com/EIPStackGroup/OpENer/files/11149086/sample_application.zip)
Hello,

 I've used the sample application as a template to create two exclusive connections with In/Out/Config assemblies. I am using a rockwell PLC and the Hilscher Ethernet/IP Tool to connect to these two connections. As long as I use just one connection, everything is working finde. But when I try to establish an additional connection ( one is active ) then the already established connection gets disconnected in the CheckAndHandleConsumingUdpSocket function (received_size == 0 or received_size == -1) and the new connection becomes active and exchanges data.

Would be great if some comes up with an idea, Thanks

OS: Windows 10 x64

### Log-Messages:

IO Exclusive Owner connection requested
No PIT segment available
networkhandler: UDP socket 536
New highest socket: 536
connection manager: connect succeeded
assembleFWDOpenResponse: sending success response
notifyMR: notify function of class 'connection manager' returned a reply
TCP reply: send 110 bytes on 260
Entering HandleDataOnTcpSocket for socket: 260
Data received on TCP: 24
Handles data for TCP socket: 260
unregister session
encap.c: Unregister Session Command
Closing TCP socket 260
networkhandler: closing socket 260
networkhandler: closing socket done 260
UDP packet to be sent to: 239.192.70.224:2222
UDP packet to be sent to: 239.192.70.224:2222
Processing UDP consuming message
**networkhandler: error on recv: 10035 - Ein nicht blockierender Socketvorgang konnte nicht sofort ausgeführt werden.**
Exclusive Owner or Input Only connection closed - Instance type: 2
Close all instance type 4 only connections
Closing UDP socket 504
networkhandler: closing socket 504
networkhandler: closing socket done 504
Closing UDP socket 504
networkhandler: closing socket 504
networkhandler: closing socket done 504
cipioconnection: CloseCommunicationChannelsAndRemoveFromActiveConnectionsList
UDP packet to be sent to: 239.192.70.224:2222
Processing UDP consuming message
Starting data length: 38
data length after sequence count: 36
Run/Idle handler: 0x1
Setting extended status: 60

### Code:

`void CheckAndHandleConsumingUdpSocket(void) {
  DoublyLinkedListNode *iterator = connection_list.first;

  CipConnectionObject *current_connection_object = NULL;

  /* see a message of the registered UDP socket has been received     */
  while(NULL != iterator) {
    current_connection_object = (CipConnectionObject *) iterator->data;
    iterator = iterator->next; /* do this at the beginning as the close function may can make the entry invalid */

    if( (kEipInvalidSocket !=
         current_connection_object->socket[kUdpCommuncationDirectionConsuming])
        && (true ==
            CheckSocketSet(current_connection_object->socket[
                             kUdpCommuncationDirectionConsuming
                           ]) ) ) {
      OPENER_TRACE_INFO("Processing UDP consuming message\n");
      struct sockaddr_in from_address = { 0 };
      socklen_t from_address_length = sizeof(from_address);
      CipOctet incoming_message[PC_OPENER_ETHERNET_BUFFER_SIZE] = { 0 };

      int received_size = recvfrom(g_network_status.udp_io_messaging, NWBUF_CAST incoming_message, sizeof(incoming_message),
        0, (struct sockaddr*) &from_address, &from_address_length);
      if(**0 == received_size**) {
        int error_code = GetSocketErrorNumber();
        char *error_message = GetErrorMessage(error_code);
        OPENER_TRACE_ERR(
          "networkhandler: socket: %d - connection closed by client: %d - %s\n",
          current_connection_object->socket[
            kUdpCommuncationDirectionConsuming],
          error_code,
          error_message);
        FreeErrorMessage(error_message);
        current_connection_object->connection_close_function(current_connection_object);
        continue;
      }

      if(####0 > received_size#####) {
        int error_code = GetSocketErrorNumber();
        char *error_message = GetErrorMessage(error_code);
        if(OPENER_SOCKET_WOULD_BLOCK == error_code) {
          return; // No fatal error, resume execution
        } OPENER_TRACE_ERR("networkhandler: error on recv: %d - %s\n",
                           error_code,
                           error_message);
        FreeErrorMessage(error_message);
        current_connection_object->connection_close_function(current_connection_object);
        continue;
      }

      HandleReceivedConnectedData(incoming_message, received_size,
                                  &from_address);

    }
  }
}`</div>

CapXilinx, 2023-04-24:
Your error is rather interesting, as the error seems after sending to a multicast group. A wireshark trace from the problematic message could help to indentify the issue.

Another topic is a different part of your error log

> Exclusive Owner or Input Only connection closed - Instance type: 2
Close all instance type 4 only connections

This should read instance type 1 and instance type 3 connection if I am not mistaken. I need to check this too in the code, if this is wrong in the repository

Suslsuk, 2023-04-24:
I created a trace at the time of activation of the second connection. Thank you

[opener_wireshark_filter.zip](https://github.com/EIPStackGroup/OpENer/files/11309900/opener_wireshark_filter.zip)

alexhaifeng, 2025-09-16:
[image]source\src\ports\generic_networkhandler.c