EIPStackGroup/OpENer #458: Implicit Messaging Error#262

MarioMay25, 2023-05-25:
Hello everyone,

I have been trying to establish communication with a 1734-AENTR that only has only one input module (1734-IV8) but I always keep getting a constant error 

[image]

This is part of the code I am using:

[image]

I used Wireshark to see where the problem was, getting the following result:

[image]

[image]

I have been trying to check if my computer ethernet network is configured to accept multicast data but have not been succesful. In addition, I am not sure if I am focusing on something that might not be the source of the problem.

I appreciate any help I can get to see if I can get this running.

Thanks everyone.

rednexela1941, 2023-05-30:
I'd look around at these. (error code `262 == 0x0106`)
```
cip/cipioconnection.c
674:          kConnectionManagerExtendedStatusCodeErrorOwnershipConflict;
682:            kConnectionManagerExtendedStatusCodeErrorOwnershipConflict;

cip/appcontype.c
199:            kConnectionManagerExtendedStatusCodeErrorOwnershipConflict;
213:            kConnectionManagerExtendedStatusCodeErrorOwnershipConflict;

cip/cipconnectionmanager.h
41:  kConnectionManagerExtendedStatusCodeErrorOwnershipConflict = 0x0106, /**< General Status has to be 0x01, Another connection has already reserved some needed resources */
```

rednexela1941, 2023-05-30:
Also, I have ran into bugs with that eeip.py library you seem to be using. Here is one fix: https://github.com/rossmann-engineering/eeip.py/pull/8.

CapXilinx, 2023-08-16:
Hi @MarioMay25 
sorry, I had to delete your last comment, as it features a screenshot from the ENIP specification.

CapXilinx, 2023-08-16:
closing also this issue, as it seems its not about OpENer, but a different library