EIPStackGroup/OpENer #459: Connection Path Question

chruetli, 2023-06-16:
Hello everyone,
I'm new to E/IP and the answer may be obvious...
I am communicating with a device where the following connection path is used:
`m_parameters.connectionPath = {0x20, 0x04, 0x24, 0x03, 0x2C, 0x64, 0x2C, 0x65}; `

How do I map this connection path to the OpENer sample application?

Thanks for your time!

CapXilinx, 2023-07-10:
Class ID 4, Instance ID 3, Member IDs 0x64 and 0x65

What kind of connection do you want?
currently the input assembly is 0x64, so the input path should be ok
If it should be an Exclusive Owner connection your output assembly needs to be changed from 151 (0x97) to 0x65
If you want an input only or listen only connection then you need to change both the output assembly and the input assembly to the corresponding heartbeat assemblies to match you EPath.

Hope that helps.

Best regards,
Martin

chruetli, 2023-07-13:
Thanks Martin

My bad, I messed up a number, but I think your statement doesn't change. The connection path is 
`{ 0x20, 0x04, 0x24, 0x01, 0x2C, 0x65, 0x2C, 0x6B }`
--> Class ID 4, Instance 1, Member IDs 0x65 and 0x6B

For the connection I set the following parameters:
- Class: 1, Trigger Type: cyclic
- T2O RPI: 500'000, T2O Connection: P2P, scheduled priority, fixed size, size = 10
- O2T RPI: 250'000, T2O Connection: P2P, scheduled priority, fixed size, size = 8

But something doesn't seem to be right
OpENer tells me:
```NotifyMessageRouter: routing unconnected message
NotifyMessageRouter: calling notify function of class 'connection manager'
notify: found instance 1
notify: calling ForwardOpen service
.../OpENer/source/src/cip/cipconnectionmanager.c: 575 (ForwardOpen),
ForwardOpen: ConConnID 0, ProdConnID 109510657, ConnSerNo 1
We have a Non-Null request
We have a Non-Matching request
.../OpENer/source/src/cip/cipconnectionmanager.c: 1343,  32 4 36 1
Received connection path size: 4
classid 4 (assembly)
Configuration instance id 1
notifyMR: notify function of class 'connection manager' returned a reply
TCP reply: send 56 bytes on 6
Entering HandleDataOnTcpSocket for socket: 6
Data received on TCP: 46
Handles data for TCP socket: 6
Send Request/Reply Data
NotifyMessageRouter: routing unconnected message
NotifyMessageRouter: calling notify function of class 'identity'
notify: found instance 1
```

And my application tells me:
```
[INFO] Send request: service=0x54 epath=[classId=6 objectId=1]
[ERROR] Message Router error=0x1 additional statuses [0x315: Invalid segment in path]
```

chruetli, 2023-07-31:
I'm don't know which message size you are thinking about. I found `set( OPENER_ETHERNET_BUFFER_SIZE "512" CACHE STRING "Number of bytes used for the Ethernet message buffer")`  or do you point to the `g_assembly_dataNNN[]` arrays?

CapXilinx, 2023-08-01:
No, I mean the Class 1 connections. OpENer only supports fixed size and the sample application/EDS uses 32 bytes for O->T and T->O directions

chruetli, 2023-08-02:
Sorry for asking again but I can only find `opener_sample_app.eds`. But I doubt that this file has any active role in the sample application. 
Can you please point me to the place where I have to set the message sizes?

CapXilinx, 2023-08-16:
Hi @chruetli 
It is not read in by OpENer, but is used by the originator usually.

MartinMelikMerkumians, 2024-01-08:
Closed due to inactivity