EIPStackGroup/OpENer #467: Need help using shared libraries.

davenmccalla-diagraph, 2023-08-02:
My goal is to build OpENer as a shared library, then write an application as a separate project that links to the shared library. I would like to start by making the sample application a separate project. Are main.c and sampleapplication.c the only source files needed for sample application? How do I determine what header files will need to be included from the OpENer project to build the sample application?

CapXilinx, 2023-09-27:
@davenmccalla-diagraph there is already a CMake switch to compile the core OpENer functionality as a shared library

davenmccalla-diagraph, 2023-09-27:
I was aware of the CMake switch and successfully built the shared libraries. The next challenge was to figure out the best way to include the OpENer header files in my project. I ended up making OpENer a submodule of my project and including these directories.

set(OpENer_INCLUDES
${CMAKE_SOURCE_DIR}/OpENer/source
${CMAKE_SOURCE_DIR}/OpENer/source/src
${CMAKE_SOURCE_DIR}/OpENer/source/src/cip
${CMAKE_SOURCE_DIR}/OpENer/source/src/enet_encap
${CMAKE_SOURCE_DIR}/OpENer/source/src/ports
${CMAKE_SOURCE_DIR}/OpENer/source/src/utils
${CMAKE_SOURCE_DIR}/OpENer/source/src/cip_objects
${CMAKE_SOURCE_DIR}/OpENer/source/src/ports/nvdata
${CMAKE_SOURCE_DIR}/OpENer/bin/posix/src/ports
${CMAKE_SOURCE_DIR}/OpENer/source/src/ports/POSIX
${CMAKE_SOURCE_DIR}/OpENer/source/src/ports/POSIX/sample_application
)

closed 2023-09-27 by davenmccalla-diagraph