EIPStackGroup/OpENer #487: OpENer/source/src/enet_encap /cpf.c   line:247

superPN, 2023-12-20:
Looks like a bug here:
We get item_count from the obtained data
Next we judge the value of item_count,
If the value of item_count is 2
The program if(common_packet_format_data->item_count >= 1U) { ... }will be executed first
Then the program if(common_packet_format_data->item_count >= 2) { ... } will be executed
When executing the if(common_packet_format_data->item_count >= 1U) program, the pointer "data" will be moved
When executing the if(common_packet_format_data->item_count >= 2) program again, the position of the "data" pointer may not be the position we want.

I feel like it should be written like this:
if(common_packet_format_data->item_count >= 2)
{
 ...
}
else if(common_packet_format_data->item_count >= 1U)
{
 ...
}

azoitl, 2023-12-20:
If we would write it like you suggested it the code in item_count >=2 would need to duplicate all the code in item >=1.

The idea in the code is to parse as many elements as there are. So the first item first and if there is a second item then the second one.

superPN, 2023-12-20:
I understand what you mean. I misunderstood. Thank you for your answer.

azoitl, 2023-12-20:
It has been a long time since we wrote that code. By now I would write it a bit different with more methods than our intention would also be clearer.

closed 2023-12-22 by superPN