EIPStackGroup/OpENer #488: OpENer/source/src/cip /cipcommon.c   link:1407

superPN, 2023-12-22:
This does not seem to match the latest standards of Ethernet/IP
in "switch(*message_runner)"

Your processing logic is like this:
If "Logical Format" is 01 (LOGICAL_SEGMENT_FORMAT_SIXTEEN_BIT)
Will execute "message_runner += 2;" and then obtain the information
That is, skip the current Logical Segment (one byte) and then skip an additional byte to obtain the data.

But in the latest Ethernert/IP standard (Vol1_3.33 page:1442), whether to skip an extra byte depends on the "Logical Segment
Type ",if"Logical Segment Type " is 111(Extended Logical), there is an extra byte to record "Extended Logical Type", and "Logical
Format" doesn't matter.

superPN, 2023-12-22:
[image]

MartinMelikMerkumians, 2024-01-08:
Thanks, I am aware that the current version is not up-to-date with the latest specification.

I deleted your image, and I need to ask you not to post screenshots from the specification, as this could lead to a takedown notice, as the specification is not an open document and copyrighted to the ODVA