EIPStackGroup/OpENer #520: STM32 Cache coherence

mulcmu, 2024-11-22:
@PeterChristen577 First off, thanks for contributing the STM32 port.  Below problem was encountered running on a H7 cpu.

I got the demo running fine on the STM32F746-Disco board.  I imported OpENer successfully into a new STM32F746-Disco TouchGFX template project.  Needed one small change for freeRTOS CMSIS_V2 vs _V1.  Both these were tested with enip/explorer and wireshark.  Explicit and implicit messaging working perfect.

I imported into a STM32H745I-Disco TouchGFX template project.  Got everything setup for this board but started getting some random data in the enip/explorer implicit messaging.  Wireshark showed a few good packets were transmitted and then one or two malformed packets would be sent.  Troubleshooting this lead to what appears to be a cache coherence issue.  

In cipioconnection.c SendConnectData(...) I changed the the outgoing message to
```
static ENIPMessage outgoing_message __attribute__((section(".outgoing_message_hack")));
```
 and stuck this with the non-cached lwip heap memory.  This got the sample application running on this board without any malformed io packets (at least for single connection).  The cpu frequency was higher on this board and the ETH is MII vs RMII which might be contributing factors.  I didn't check the F746 to see if malformed packets were still occurring but just way less frequently.

Any ideas for a more elegant solution than the above hack?

PeterChristen577, 2024-11-27:
Hi mulcmu

Same questions/ideas:

- Which heap size have you configured in FreeRTOSConfig.h?
- What is the MEM_SIZE in lwiopts.h
- Did you have a look to the document [OpENer STM32 Port.pdf](https://github.com/EIPStackGroup/OpENer/blob/master/source/doc/STM32/OpENer%20STM32%20Port.pdf)? 
There are some hint for setup of lwiopts.h

Best regards Peter

mulcmu, 2024-11-28:
The freeRTOS heap size is `#define configTOTAL_HEAP_SIZE                    ((size_t)0x8000)`

lwip heap is 96k.  First 32k of D2 sram is used for Ethernet dma descriptors and rx buffers, rest of the 128k is used for lwip.
`#define MEM_SIZE 0x18000`

The graphics buffers are in the external 16mb SDRAM.

Here's snapshot of the stack sizes.  OpENer was already increased to (4*1024) as well as increases to the default lwip tasks.
[image]

PeterChristen577, 2024-11-28:
Everything looks good in the stack snapshot, let me do a similar analysis on the STM32F746 target.
One question: Which cycle time have you set up in enip/explorer? Is the rate of malformed packet related to the cycle time?

mulcmu, 2024-11-29:
Cycle time was the default 200ms.  At 200ms the malformed packets appeared randomly distributed, I didn't see any patterns.

This [post](https://community.st.com/t5/stm32-mcus-embedded-software/lwip-sendto-is-not-sending-the-data-in-the-buffer-passed-to-it/m-p/713278/highlight/true#M50823) on ST community forum discusses similar problem.  Seems that on my H745, sometimes the processor cached data is written out to the memory before the DMA reads it for the valid packets.  For the malformed packets the outgoing message buffer data is only in the MCU cache, so the DMA is copying the SRAM memory that holds random data.

https://github.com/EIPStackGroup/OpENer/blob/70d6947c6ef97c5c8ccbc908b94651432e18cf9c/source/src/ports/generic_networkhandler.c#L682-L719

I reverted my prior hack and then on line 693 of the SendUdpData() an explicit cache flush was added.
```
  SCB_CleanDCache_by_Addr(outgoing_message->message_buffer, sizeof(ENIPMessage));
```
This change also resolved the malformed packet issue for the implicit messaging.

PeterChristen577, 2024-12-02:
Great, that you found out this so quickly!
If I understand correctly the situation this cache flush is needed only for H7 targets and not for F7, isn't it?

mulcmu, 2024-12-03:
I was still working through things, I have 2 H7 (H757 & H745), an H563, and the F746.  The clean cache() function doesn't seem to hurt the F746 but there were a few times ping, http, and list identity would work but then the rest of the ethernet/IP appeared with errors.  Still trying to figure out what was changing.  I started a branch with the changes.  https://github.com/mulcmu/OpENer/tree/stm32-flush-cache-CMSISv2.