EIPStackGroup/OpENer #523: Multiple High-Severity Vulnerabilities in OpENer 1.5.2 Affecting Transitive Dependencies

jendcruz22, 2025-03-11:
### Summary
The npm package opener@1.5.2 includes vulnerable code derived from the EIP Stack Group OpENer implementation. Multiple CVEs have been assigned to vulnerabilities in this code base, with severities ranging from HIGH (7.5) to CRITICAL (9.8). These vulnerabilities can lead to out-of-bounds writes, use-of-uninitialized pointers, and denial-of-service (DoS) conditions.

### Details
The identified vulnerabilities in the underlying OpENer code include:

**CVE-2022-43604 & CVE-2022-43605**

Out-of-bounds write vulnerabilities in the GetAttributeList and SetAttributeList attribute_count_request functionalities. An attacker can send specially crafted EtherNet/IP requests to trigger an out-of-bounds write, potentially crashing the server or enabling remote code execution.

**CVE-2021-27478, CVE-2021-27482, CVE-2021-27498, and CVE-2021-27500**
Vulnerabilities that can lead to denial-of-service conditions by triggering crashes via specifically crafted packets sent to the vulnerable EtherNet/IP handling code.

**CVE-2022-43606**
A use-of-uninitialized-pointer vulnerability in the Forward Open connection_management_entry functionality, which can also lead to a server crash.

These issues have been documented with high or critical CVSS scores (up to 9.8), indicating severe potential impacts such as remote code execution, arbitrary data read, or complete denial-of-service.

### Impact
If an attacker is able to interact with or trigger the vulnerable code paths (e.g., through maliciously crafted EtherNet/IP requests), it may be possible to:

- Crash the server (DoS attack)
- Execute arbitrary code remotely
- Read arbitrary data from the server

Given the severity of these vulnerabilities (with CVSS scores reaching 9.8), the risk is considered high, especially if the affected functionality is accessible in any way.

### Recommendation
Please update the vulnerable dependency by either patching the underlying OpENer code or releasing a new version of the opener package that incorporates the fixes provided in newer, secure versions of OpENer.

MartinMelikMerkumians, 2025-03-11:
I am not sure if you refer to the npm package or the lastest code available here?

As I do not maintain the npm package nor initially create it, I don't have any possibly to update it. Please contact the maintainer of the npm package to update it

jendcruz22, 2025-03-27:
This vulnerability was identified for opener@1.5.2 by the OWASP tool for this packages being used by @strapi/strapi@5.12.0

@MartinMelikMerkumians would you happen to know who that is?

jendcruz22, 2025-03-27:
@MartinMelikMerkumians This vulnerability arises from https://github.com/domenic/opener. I'll submit an issue on there. Thank you.

jendcruz22, 2025-03-27:
@MartinMelikMerkumians [this issue](https://github.com/domenic/opener/issues/38) redirected me to create this issue with OpENer.

Could you please help me with who are the maintainers and how I can get in touch with them

jendcruz22, 2025-03-27:
@MartinMelikMerkumians https://github.com/domenic/opener/issues/38 redirected me to create this issue with OpENer.

Could you please help me with who are the maintainers and how I can get in touch with them

elupus, 2025-05-22:
From what i can tell: CVE-2022-43604 ,CVE-2022-43605 looks to be resolved in OpENer a while back. Like was mentioned in the references issue on domenic/opener, the scanner seem confused about what package is in use.

That package does not even use OpENer Ethernet/IP stack, so the reports are invalid on that package.