EIPStackGroup/OpENer #526: The error log always appears: too large packet received...

SupermanOfHeiLinPu, 2025-03-21:
The error log always appears "too large packet received will be ignored, will drop the data".

I modified the `sampleapplication.c` file under POSIX as follows, and I modified the value of OPENER_CIP_NUM_EXLUSIVE_OWNER_CONNS in opener_user_conf.h to 50. When I run it, I always get the error: "too large packet received will be ignored, will drop the data".

My sampleapplication.c file :
```
/*******************************************************************************
 * Copyright (c) 2012, Rockwell Automation, Inc.
 * All rights reserved.
 *
 ******************************************************************************/

#include <string.h>
#include <stdlib.h>
#include <stdbool.h>

#include "opener_api.h"
#include "appcontype.h"
#include "trace.h"
#include "cipidentity.h"
#include "ciptcpipinterface.h"
#include "cipqos.h"
#include "nvdata.h"
#if defined(OPENER_ETHLINK_CNTRS_ENABLE) && 0 != OPENER_ETHLINK_CNTRS_ENABLE
  #include "cipethernetlink.h"
  #include "ethlinkcbs.h"
#endif

// #define DEMO_APP_INPUT_ASSEMBLY_NUM                100 //0x064
// #define DEMO_APP_OUTPUT_ASSEMBLY_NUM               150 //0x096
// #define DEMO_APP_CONFIG_ASSEMBLY_NUM               151 //0x097
// #define DEMO_APP_HEARTBEAT_INPUT_ONLY_ASSEMBLY_NUM  152 //0x098
// #define DEMO_APP_HEARTBEAT_LISTEN_ONLY_ASSEMBLY_NUM 153 //0x099
// #define DEMO_APP_EXPLICT_ASSEMBLY_NUM              154 //0x09A

/* global variables for demo application (4 assembly data fields)  ************/

// EipUint8 g_assembly_data064[32]; /* Input */
// EipUint8 g_assembly_data096[32]; /* Output */
// EipUint8 g_assembly_data097[10]; /* Config */
// EipUint8 g_assembly_data09A[32]; /* Explicit */

EipUint8 g_assembly_data_temp[4096];
EipUint8 g_assembly_config[10]; /* Config */

/* local functions */

enum AppAssemblyNum {
  CONFIG_ASSEMBLY_NUM = 161,//0xA1
  INPUT_ASSEMBLY_NUM = 101, //0x65
  INPUT_ROBOT_STATE_ASSEMBLY_NUM,
  INPUT_BOARD_IO_ASSEMBLY_NUM,
  INPUT_TOOL_ASSEMBLY_NUM,
  INPUT_JOINT_ASSEMBLY_NUM,
  INPUT_TCP_ASSEMBLY_NUM,
  INPUT_BOOL_ASSEMBLY_NUM,
  INPUT_INT_ASSEMBLY_NUM,
  INPUT_FLOAT_ASSEMBLY_NUM,
  INPUT_IO_ASSEMBLY_NUM, // TODO: 删除此项
  OUTPUT_ASSEMBLY_NUM = 131, // 0X83
  OUTPUT_ROBOT_STATE_ASSEMBLY_NUM, // TODO: 删除此项
  OUTPUT_BOARD_IO_ASSEMBLY_NUM, // TODO: 删除此项
  OUTPUT_TOOL_ASSEMBLY_NUM, // TODO: 删除此项
  OUTPUT_JOINT_ASSEMBLY_NUM, // TODO: 删除此项
  OUTPUT_TCP_ASSEMBLY_NUM, // TODO: 删除此项
  OUTPUT_BOOL_ASSEMBLY_NUM,
  OUTPUT_INT_ASSEMBLY_NUM,
  OUTPUT_FLOAT_ASSEMBLY_NUM,
  OUTPUT_ROBOT_IO_ASSEMBLY_NUM
};
[... 264 lines cut ...]
          size_t size_of_element) {
  return calloc(number_of_elements, size_of_element);
}

void CipFree(void *data) {
  free(data);
}

void RunIdleChanged(EipUint32 run_idle_value) {
  OPENER_TRACE_INFO("Run/Idle handler triggered\n");
  if( (0x0001 & run_idle_value) == 1 ) {
    CipIdentitySetExtendedDeviceStatus(kAtLeastOneIoConnectionInRunMode);
  } else {
    CipIdentitySetExtendedDeviceStatus(
      kAtLeastOneIoConnectionEstablishedAllInIdleMode);
  }
  (void) run_idle_value;
}

```

Through debugging and packet capturing, I've found that in the `NetworkHandlerProcessCyclic()` function of the `generic_networkhandler.c` file, the call to `HandleDataOnTcpSocket()` receives UDP data on port 2222 and parses this packet in an incorrect way.

[image]
(I added some data printing to the error log.)

alexhaifeng, 2025-08-25:
修改文件 sorce/CMakelist.txt 第72行:
#add_definitions(-DPC_OPENER_ETHERNET_BUFFER_SIZE=${OPENER_ETHERNET_BUFFER_SIZE} )
add_definitions(-DPC_OPENER_ETHERNET_BUFFER_SIZE=1024)
我也遇到同样的问题，已经解决了，你可以试试，问题已经提了五个月，希望能帮到你，哈哈哈

SupermanOfHeiLinPu, 2025-08-26:
@alexhaifeng 哈哈，谢谢，我试试

alexhaifeng, 2025-09-05:
@SupermanOfHeiLinPu 问一下，看你的代码，有多个输入和输出组件，是通过建立多个连接来实现的吗，一个连接是不是只能一个输入一个输出？

alexhaifeng, 2025-09-16:
> [@SupermanOfHeiLinPu](https://github.com/SupermanOfHeiLinPu) 问一下，看你的代码，有多个输入和输出组件，是通过建立多个连接来实现的吗，一个连接是不是只能一个输入一个输出？

这个问题已经解决了，确实是一个独占连接只能一个输入组件一个输出组件